FreeOpcUa/python-opcua #491: How to get multiple node values

luchs3, 2017-08-27:
Hi,
How is it possible to pass a list of Node IDs and get their Values with one request. I only found the possibility to get one Value at a time.
Thanks a lot for your suggestions

oroulet, 2017-08-27:
The idea is that you copy code of get_value and modify it to your need. Should be easy

luchs3, 2017-08-28:
That works very well of course. My challenge is to get many values at a quite high frequency (every 1-2ms). That works with 2 Nodes in a loop but the loop process time rises linear with every additional Node. My thought was, to put every request in a separate Thread so maybe it becomes faster. But if the Server can send a bunch of Node Values at once it would be much more elegant. At the C++ Stack you can pass a list of Nodes and get its values back. But I don't know how it works in the back and if the UA specifications support such a feature at all.

oroulet, 2017-08-28:
If you look at code of get_value, you will see that you can read value of
several nodes at once

luchs3, 2017-08-28:
Perfect! That works great.
Thanks for your help!

closed 2017-08-28 by luchs3

amisha218, 2019-06-11:
I am having the same issue. I want to get the multiple values at a time but I am facing problem in this.
Get_value is giving error: list object has no attribute 'get_value'.
Somebody please help!

oroulet, 2019-06-11:
@amisha218 stopp hijacking other bug request. And in that case you need to learn more python before asking