FreeOpcUa/python-opcua #492: MaxMessageSize and MaxChunkCount set to 0

jwbeck, 2017-09-06:
In freeopcua/ua/uaprotocol_hand.py, the Hello class initializes the max message size and max chunk count to 0. This made my client script not work with the server I was using. After changing these values to match what I was seeing with UaExpert (MaxMessageSize = 16777216 and MaxChunkCount = 5000), my script worked.

oroulet, 2017-09-07:
0 is definitely the correct value as spec. But some servers/clients are broken

jwbeck, 2017-09-07:
The part where it would fail was when I would do create_session() or get_endpoints(), my server would return an error, and I would get a timeout error in the script.

daniel-brosche, 2017-11-03:
I had also problems with the values with the Unified Automation HP Server. It should be possible to configure these values. So it is more flexible in cases of embedded ressource restricted environments.
As far as I understand the code it should be not complicated to pass this parameters. Could be realized with 0 as default parameter.

From Part 6:

> The maximum size for any response Message. The Server shall abort the Message
> with a Bad_ResponseTooLarge StatusCode if a response Message exceeds this
> value.
> The mechanism for aborting Messages is described fully in 6.7.3.
> The Message size is calculated using the unencrypted Message body.
> 
Later on in the Spec:
> A value of zero indicates that the Client has no limit.

Therefore 0 is a valid value but not every server accepts this value. I see benefit to make this parameter configurable.

oroulet, 2017-11-03:
maybe add it as argument to hello method to Uaclient class. Then people can call UaClient.hello(whatever) directly

daniel-brosche, 2017-11-03:
I have associated this line with the issue:

https://github.com/FreeOpcUa/python-opcua/blob/10a9064b2e2d6ec1b63cad1bff398d175d026cb6/opcua/client/client.py#L358

Is this correct?

jwbeck, 2017-11-03:
https://github.com/FreeOpcUa/python-opcua/blob/f935c993dc8fd4f650b71ebc33dacf192dc6af1f/opcua/ua/uaprotocol_hand.py#L20

oroulet, 2017-11-03:
yes make a PR with a very high size (Check what value uaexpert use for example) and add possibility to pass it as arg to UaClient.hello() method

daniel-brosche, 2017-11-03:
Ok, I will do that :+1:

closed 2021-01-30 by jwbeck