FreeOpcUa/python-opcua #493: opc ua server in siemens cpu

lucafei, 2017-09-06:
hello, guys,

i want to ask, have you tried to use 4diac to communicate with opc ua server in cpu of siemens?

oroulet, 2017-09-07:
What do you mean? 4diac seems to iuse open62551 for it's it UA support. So
we should be able to communicate with it

oroulet, 2017-09-07:
but it is not the first time a server is not happy with 0 values, maybe we should set them to something else to avoid troubles..

lucafei, 2017-09-07:
i have used 4diac client to communicate with opc ua server in cpu of siemens, but it doesn't work, because  4diac client supports only method call and siemens opc ua server supports only variable as Nodeclass. So i want to ask you if you tried to use 4diac communicate with iopc ua server in cpu of siemens.

zerox1212, 2017-09-07:
You should probably ask the open62551 devs. When I looked at 4DIAC quickly it says that it supports variable nodes.

oroulet, 2017-09-09:
what does testing a 4diac client with an siemens server has to do with python-opcua?

closed 2017-09-09 by oroulet