FreeOpcUa/python-opcua #495: Using opcua-client in raspberry pi

abhinavrawat27, 2017-09-09:
Hi all

I have raspberry pi 3 and I have installed opcua-client gui in it.

`sudo pip3 install opcua-client`
 `sudo pip3 install cryptography`
 `sudo apt-get install python3-pyqt5`

I am able to run opcua-client gui. Before testing it on pi, I tested the server using uaexpert on windows and I was able to connect and get data from the server. in opcua- client gui, I entered server IP Address

`opc.tcp://192.168.111.231` and tried to connect but it gave below error:

`uaclient.uaclient - INFO - Connecting to opc.tcp://192.168.111.231/ with parameters None, None, None, None')`
`uaclient.mainwindow - WARNING - showing error: %s')`
`uawidgets.utils - ERROR - [Errno 111] Connection refused')`
`Traceback (most recent call last):`
 ` File "/usr/local/lib/python3.4/dist-packages/uawidgets/utils.py", line 21, in wrapper`
    `result = func(self, *args)`
  `File "/usr/local/lib/python3.4/dist-packages/uaclient/mainwindow.py", line 326, in connect`
    `self.uaclient.connect(uri)`
  `File "/usr/local/lib/python3.4/dist-packages/uaclient/uaclient.py", line 86, in connect`
    `self.client.connect()`
  `File "/usr/local/lib/python3.4/dist-packages/opcua/client/client.py", line 243, in connect`
    `self.connect_socket()`
  `File "/usr/local/lib/python3.4/dist-packages/opcua/client/client.py", line 262, in connect_socket`
    `self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)`
  `File "/usr/local/lib/python3.4/dist-packages/opcua/client/ua_client.py", line 215, in connect_socket`
    `return self._uasocket.connect_socket(host, port)`
  `File "/usr/local/lib/python3.4/dist-packages/opcua/client/ua_client.py", line 135, in connect_socket`
    `sock = socket.create_connection((host, port))`
  `File "/usr/lib/python3.4/socket.py", line 509, in create_connection`
    `raise err`
 ` File "/usr/lib/python3.4/socket.py", line 500, in create_connection`
    `sock.connect(sa)`
`ConnectionRefusedError: [Errno 111] Connection refused`
`uaclient.mainwindow - WARNING - showing error: %s')`

I want to know why the error says Connecting to opc.tcp://192.168.111.231/ with parameters None, None, None. While using UAExpert on windows, it didnt required any port number. So I didnt used any port number.

Thanks.

closed 2017-09-09 by abhinavrawat27

mohammad1700, 2018-07-12:
Hi abhinavrawat27,
Have you managed to solve this issue?

abhinavrawat27, 2018-07-12:
@mohammad1700  yes I was missing the port number in opcua-client gui. So you enter the correct IP address with the port in order to access the server.

KarenSimonyan, 2019-05-14:
Here is the solution:
sudo apt-get install python3-lxml python-lxml libxml2-dev libxslt-dev python-dev

FabioAlmeida25, 2021-06-24:
Hello guys

FabioAlmeida25, 2021-06-24:
Can anyone help me, please?
[image]
[image]

swamper123, 2021-06-28:
@FabioAlmeida25 Please open a new discussion with some backround information.

ArmanAttar, 2022-11-21:
Hi @FabioAlmeida25 Have you solved the mentioned issue? I got the same problem

munishreddy1, 2023-02-23:
When I am trying to install opcua-client in my raspberry pi with armv7 architecture while installing the Pyqt5 module it is throwing the below error. I have seen in some forums it is because of the python version. So I try to create a docker image of my python script with the python 3.9 version along with all dependencies and in armv7 architecture. While building the docker image it is throwing the same error while installing the pytq5 module. When I try to install the packages in amd64 architecture PCs it is working fine. Can anyone please help me with this?

 error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [30 lines of output]
      Querying qmake about your Qt installation...
      Traceback (most recent call last):
        File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 585, in setup
          self.apply_user_defaults(tool)
        File "/tmp/pip-install-n7wvxyan/pyqt5_eecbbf17d381465997ddd8547dd4cb77/project.py", line 68, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 236, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/pyqtbuild/builder.py", line 78, in apply_user_defaults
          self._get_qt_configuration()
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/pyqtbuild/builder.py", line 493, in _get_qt_configuration
          for line in project.read_command_pipe([self.qmake, '-query']):
        File "/tmp/pip-build-env-b5pupna4/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 553, in read_command_pipe
          raise UserException(
      sipbuild.exceptions.UserException
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.

AndreasHeine, 2023-02-23:
python-opcua is for python 3.6 and below and is not maintained anymore...

use asyncua (opcua-asyncio) instead!!!

your error on the first look has nothing to do with opcua...

munishreddy1, 2023-02-24:
Thanks, @AndreasHeine for your reply. Sure I will use opcua-asyncio. But I am facing a problem while installing the PyQt5 package required opcua-client gui. Even with asyncio, I need to install the client in pi to connect to my server. Please correct me if I am wrong I am new to opcua. Thanks again for your help.

AndreasHeine, 2023-02-24:
@munishreddy1 this is probalby the better place to post your issue: https://github.com/FreeOpcUa/opcua-client-gui

this is an closed issue from 2017 posting on closed issues is kind of odd on github... 
well aged issue i would say ;)