FreeOpcUa/python-opcua #496: How to access variables using node id ?

SAndrew27, 2017-09-10:
I have an `OPCUA server` and I am trying to connect to it. I have successfully installed its opcua client gui and able to browse to it.This is how the address space looks like:

   
    Root                         0:Root                          i=84
    |->Objects                   0:Objects                       i=85
      |->Server                  0:Server                        i=2253
      |->Data Set                2:Data Set                      ns=2,i=50001
      |->PLC                     4:PLC                           ns=4,i=20000
       |->Modules                6:Modules                       ns=6,i=1000
         |->Data PV              6:Data PV                       ns=6,s=::AsDataPV
           |->Var1               6:Var1                          ns=6,s=::AsDataPV:Var1
           |->Var2               6:Var2                          ns=6,s=::AsDataPV:Var2
           |->Var3               6:Var3                          ns=6,s=::AsDataPV:Var3
           |->Var4               6:Var4                          ns=6,s=::AsDataPV:Var4
           |->Var5               6:Var5                          ns=6,s=::AsDataPV:Var5

I want to get the values of `var1, var2, var3 ,var4, var5`. How can I extract its values by using their node ids.

Thanks

zerox1212, 2017-09-10:
What do you mean extract? The client GUI displays variables and their values. If you want to get the variable values and do something with them you will have to use the client library. There are many examples for this.

SAndrew27, 2017-09-10:
Yes how to use the library to extract the values. Can you please show me an example

zerox1212, 2017-09-10:
Please take a look at the repo and at least try to make it work on your own before making an issue.

https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-minimal.py

SAndrew27, 2017-09-10:
HI
I am able to get the root node and its children too.

client = Client("opc.tcp://192.168.111.231:4840")
client.connect()
root = client.get_root_node()
print(root.get_children())

This print out the its children with node it i=84,i=85,i=86. I want to know How to proceed further and access the node using i=84.

zerox1212, 2017-09-10:
You can keep calling `x.get_children()` on child nodes until you have the list of nodes you want. Or you can get a node directly by it's ID as shown in the examples.

`var = client.get_node("ns=6;s=::AsDataPV:Var1")`

amisha218, 2019-06-06:
how to write values from your python client to opc scout?

oroulet, 2019-06-06:
@amisha218 please stop hijacking old bug reports. If you have a question, create a new bug report and give enough information. Read this for example: https://github.com/qTox/qTox/wiki/Writing-Useful-Bug-Reports

closed 2019-06-06 by oroulet