FreeOpcUa/python-opcua #497: Error: An attempt was made to access a socket in a way forbidden by its access permissions

abhinavrawat27, 2017-09-11:
Hi 

I am running server_minimal example. I am using python2.7. But when I am running the example it shows below error:

Traceback (most recent call last):
  File "E:\Workspace\OPCUAServer\server.py", line 28, in <module>
    server.start()
  File "C:\Python27\lib\site-packages\opcua\server\server.py", line 277, in start
    self.bserver.start()
  File "C:\Python27\lib\site-packages\opcua\server\binary_server_asyncio.py", line 97, in start
    self._server = self.loop.run_coro_and_wait(coro)
  File "C:\Python27\lib\site-packages\opcua\common\utils.py", line 193, in run_coro_and_wait
    return task.result()
  File "C:\Python27\lib\site-packages\trollius\futures.py", line 285, in result
    compat.reraise(type(self._exception), self._exception, exc_tb)
  File "C:\Python27\lib\site-packages\trollius\tasks.py", line 259, in _step
    result = coro.send(value)
  File "C:\Python27\lib\site-packages\trollius\base_events.py", line 861, in create_server
    % (sa, err.strerror.lower()))
socket.error: [Errno 10013] error while attempting to bind on address ('0.0.0.0', 4840): an attempt was made to access a socket in a way forbidden by its access permissions

greatdsa, 2018-03-07:
It's due to the windows firewall or antivirus. Try switching them off and then try again..

letsgogeeky, 2018-03-22:
It's just that you're trying to use a forbidden port. Just try to change to port **9999** for example. 

in your main function:
`if __name__ ==  "__main__":
    app.run(port=9999)`

xsqian, 2019-07-14:
What is specific about port 4840? 
I got this error message when run on port 4840:
OSError: [Errno 10013] error while attempting to bind on address ('127.0.0.1', 4840): an attempt was made to access a socket in a way forbidden by its access permissions
but runs fine on port 4841.
any help will be appreciated!