FreeOpcUa/python-opcua #498: ImportError: cannot import name Client

abhinavrawat27, 2017-09-13:
Hi all

I have installed opcua on my windows. I am using python 2.7 and pydev in eclipse. I have created a project `OPCUAServer` to test the server example. It worked fine. Then I created a project `OPCUAClient` to test the client example but it gave me the import error.
As I have installed the opcua so I do not need to use `import sys
sys.path.insert(0, "..")` . In my OPCUAServer project I normally import the server using

`from opcua import Server`

but when I am doing the same in Client project `from opcua import Client`, it is giving me the below error:

`Traceback (most recent call last):`
 ` File "E:\Workspace\OPCUAClient\main.py", line 2, in <module>`
    `from opcua import Client`
  `File "E:\Workspace\OPCUAClient\opcua.py", line 10, in <module>`
`ImportError: cannot import name Client`

Now the interesting part is when I included the Client in my server project:

`from opcua import Server`
`from opcua import Client`

It worked fine. 
Can anyone please tell me whats wrong I am doing.

Thanks
Abhinav

closed 2017-09-28 by abhinavrawat27

MartinTrat, 2017-10-20:
Same Error, can someone please explain?

abhinavrawat27, 2017-10-20:
Hi MartinTrat

I was facing this problem because I had few files and directory issues. 
Are you sure you have installed freeopcua.
I have closed this topic. You can open another one with exact error you are facing.?

seemas7, 2019-10-12:
Hi Abhinav,

I am still facing the problem.May i know what files and directory cause the issue?

oroulet, 2019-10-12:
This is an installation issue. Either you have several python versions installed and rub the wrong one or the installation completely failed for some reasons. You should ask for help in some python forums. This had nothing to do with opcua

seemas7, 2019-10-12:
I have just one python version installed ..   I m trying this on RPI as
well as pycharm on windows which gives me the same error

OptrixAU, 2020-02-10:
One thing that can cause this error is if you've named YOUR python file **opcua.py**

In this case, it will look for the 'clients' class inside your own file, not inside the actual opcua library namespace. 

If so, rename your Python script, and ensure there are no .pyc files in the directory still called 'opcua.pyc'.