FreeOpcUa/python-opcua #499: python opcua connection

lasdarf, 2017-09-15:
Hi everyone,

I am writing to you, because I am desperate. I need to start an opc ua server on one pc, load variables and read them with an opc ua client on the other pc. 

I tried a simple python code, but I cannot read my variables. I just declared them like this:

uri = "Lasdarf" %%% random name here?
idx = server.register_namespace
myobj = objects.add_object(idx, "MyObject")
myvar=obj.add_variable(idx, "MyVariable", 6.7) 

If I take as a node address something like ns=1;i=2002, (I get this if I type myvar in python), my client doesn't find its value. Is there anything else, a specific structure I need to give my nodes/variables? Do I need to somehow "upload" or "send" them? Is this even the right NodeID, or do I need to give something else? 

It would be so great, if you could help me out how I get my correct IP addresses and Node IDs, or just let me know of a simple python program exists that only opens a server, inititalizes a variable with a number value and "saves" it in a node and gives me the correct addresses I need to put into the client...

You see the tragedy of my problem ;) PLEASE!

Thanks, 

Lasdarf

oroulet, 2017-09-15:
First startwith one of the examples under examples. they should all work. then tell us what client do you use and copy the entire error message. If you use a strange client then start with a common one prosys/uaexpert or python-opcua

lasdarf, 2017-09-15:
Thanks for answering.

I use the python example, "Server_minimal.py", which is working fine in Python. I use the following code:
    
   # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0:4840")

    # setup our own namespace, not really necessary but should as spec
    uri = "ADI"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()

    # populating our address space
    myobj = objects.add_object(idx, "MyObject")
    myvar = myobj.add_variable(idx, "MyVariable", 6.7)
    myvar.set_writable()    # Set MyVariable to be writable by clients

    # starting!
    server.start()_

I can get my server setup and I can also find the server, if I have the client (UaExpert or the Python GUI) on the same computer and using localhost. This is working fine. 

Now I want to connect PC1 with PC2 using opcua. I run the same server example, which I used before on PC1 and now I start UaExpert Client on PC2. 
I type in the endpoint URL, I set in the Python_example, and I get this error in the UaExpert Client on PC2: "Could not connect to server:BadCommunicationError"

I tested it and the server is alive and I can connect to it on the same machine (PC1). I can see it completely, but I cannot connect to the server on PC2.

I think I'm doing something very basic wrong.

Thanks,

Lasdarf

oroulet, 2017-09-15:
server.set_endpoint("opc.tcp://0.0.0:4840")

First this line is missing a 0. Then using 0.0.0.0 does not work on
OS, replace it with your IP or machine name. If you still have issues
then you gave a firewall somewhere inbetween

lasdarf, 2017-09-15:
Thanks for your help - heavily appreciated. I established the connection between the two PCs and I can now communicate.

closed 2017-09-15 by lasdarf