FreeOpcUa/python-opcua #501: Help with basics

GregorPi, 2017-10-12:
Hi,

having to implement my first real project on a Raspberry Pi, I am having problems with basic knowledge. I want to set up an OPC UA Server on an Raspberry Pi and read in two sensors with it. The data from these sensors shall be sent out via the OPC UA Protocol. On recommendation, I want to use the python-opcua server. 
I already secessfully managed to install the package and all the other packages needed fpr the python-opcua too.
Now I want to use the server-minimal.py File to take the first steps into this new topic.

Unfortunately I do not know the needed instructions I have to put into the terminal to setup the server and finally start it.

I would be glad, if someone could take the time to give me a short explanation of how to set up an OPC UA Server

Thanks,
Gregor

zerox1212, 2017-10-12:
You can either run it from an IDE ( I think the Pi comes with one installed) or you can open a terminal and do `python /path/server-minimal.py` replacing "path" with the location of your .py file. If you don't want to type the path you can use the `cd` command to change directories until you are in the folder that has your .py file, then you can do `python server-minimal.py` and it will run.

GregorPi, 2017-10-17:
Hi,

thanks for your response. It seems like something went wrong on my first installation. Therefore, I formarted the SD-Card and reinstalled Raspbian.
For anyone that is interested, as the documentation is very poor:
Following commands were put in the  terminal:

sudo apt-get update
sudo apt-get upgrade

sudo pip install cryptography
sudo pip install python-dateutil
sudo apt-get install python-lxml
sudo pip install pytz
sudo pip install enum34
sudo pip install trollius
sudo pip install futures

sudo pip install freeopcua

git clone https://github.com/FreeOpcUa/python-opcua
cd python-opcua
cd examples
python server-minimal.py

To connect to the server with a client, the following adress has to be used:
opc.tcp://IP-adress of your Raspberry:4840

zerox1212, 2017-10-17:
So you know, you don't need to `pip install freeopcua` if you are going to clone the master from GitHub.

Also you might want to look at changing to Python 3, the Pi probably defaults to using Python 2.7, but Python 3 is preferred because some 2.7 libraries (like trollius) are no longer maintained.