FreeOpcUa/python-opcua #503: basic client functionality

Boy1995, 2017-10-24:
Which services do you need for a basic OPC UA Client (only OPC classic DA functionality)

zerox1212, 2017-10-24:
What is your use case? I don't understand the question. You want only read and write of variables and arrays? You want to remove all the code for running a UA server?

Boy1995, 2017-10-24:
Yes i would like to only read and write only variables.

oroulet, 2017-10-24:
OPC da is not the same as ua. Entirely different beast

Boy1995, 2017-10-24:
I understand, but a few functions in UA are the same as opc DA and i would like to try to build that functionality.

oroulet, 2017-10-24:
Da is an entirely different protocol based on Microsoft dcom. Ua is a
custom protocol. There is no similarities apart from name and field of use

Boy1995, 2017-10-24:
I just mean the ability to read and write data from the server and only read the real-time data

zerox1212, 2017-10-24:
You do not need any services... You just need Python and this library. Then you can either make a server that allows read/write, or make a client and connect to a server and read/write.

Boy1995, 2017-10-25:
But i want to convert it to a PLC

zerox1212, 2017-10-25:
You will only be able to connect it to a PLC that has an OPC UA server built in. This doesn't include any proprietary PLC protocols.