FreeOpcUa/python-opcua #504: Installing using anaconda

EnzoOestanto, 2017-10-25:
Hi, I'm very new to Python and Anaconda. after a couple hours trying to install the library, I still not sure how to install it. can we install using conda install or pip and how exactly to do that? I'm currently using anaconda 1.6.9 in windows. sorry if this might sound like a silly question

oroulet, 2017-10-25:
After installing anaconda. Within anaconda you can probably start a shell or console and write: pip install opcua or maybe pip install freeopcua

tdesroches, 2017-10-25:
I've never used conda but you can install it through pip under the package name freeopcua

zerox1212, 2017-10-25:
Is there a reason you need to use Anaconda? I always found using normal Python and pip to install what I need a lot easier.

EnzoOestanto, 2017-10-25:
thanks for the reply form everyone, I eventually managed to install it using "pip install freeopcua" and i also just realized that pip is run from the command line, not the Python interpreter. thanks for the help.

@zerox1212 , I read some articles and thought that by using anaconda, i can manage the dependencies easier. but, i am also very new to this, so i don't know much

zerox1212, 2017-10-25:
I started with Python the same way. Then I quickly realized that it only made it more confusing. The fact that you used pip and know what dependencies are means you should probably not bother with bloated software like Anaconda, because the reverse will happen. Anaconda comes with so many libraries already installed it's hard to know if Anaconda has it or not.

Since you are on windows I recommend just grabbing the base python installation https://www.python.org/downloads/ . Then installing PyCharm as your IDE https://www.jetbrains.com/pycharm/ . Then you can use PyCharm to manage your packages.
[image]
(note I'm on a corporate network so the search is blocked, but normally you can find all python packages directly from the IDE and you don't need to use a shell)

EnzoOestanto, 2017-11-02:
@zerox1212 , thanks for the recommendation. I also just read about PyChar. It seemed like a great IDE. The same developer also developed Android Studio (which I am currently also using to develop android app). they look very similar. from my experience in android studio, I can see that it is a great IDE. I would definitely try it out. Thanks man.

harua8n, 2017-12-29:
I use miniconda, then do:
conda update --all
pip install freeopcua