FreeOpcUa/python-opcua #506: Nodes with fixed nodeids

ablakey, 2017-11-02:
Is it possible (or even reasonable) to have fixed node ids so that I can help guarantee a nodeid path is the same across reboots/configuration changes?  I even noticed there's a concept of nodeid paths being string-based rather than integer based? http://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l2_ua_node_ids.html

I'm having a problem with OPC software (Ignition) that wants to reference all nodes by ID rather than browser path. If the OPC server is restarted and a different number of nodes are launched, their Ids all change and now none of the Ignition tags are correct anymore.

oroulet, 2017-11-02:
When creating a new node you can give a node Id as argument. Then you have
fixed nodeid

ablakey, 2017-11-02:
Thanks for the quick response. I'll have to re-review the workflow I use to create these nodes/variables. 

Maybe this is a lack of understanding on my part.  I have a node, which I found using its browse name and add a "variable" to it. But that "variable" is basically a node, too? If viewing my structure using opcua-client-gui I see that each "variable" has a nodeid path, which are autogenerated: 

```
node = self._server.nodes.objects.get_child(['2:System', '2:SystemMonitor'])
node.add_variable(self._base_node_id, 'Heartbeat', 0)
```

[image]

Mostly just musing out loud. I don't expect free support on this; I can keep digging myself.

oroulet, 2017-11-02:
Yes everything is a node. In this case a node of type variable

zerox1212, 2017-11-02:
If you don't give an explicit node id when you create the node (variable in your case) the system will auto create one.

I personally prefer to design the address space with XML. Then you can make it exactly like you want and ids will be fixed.

ablakey, 2017-11-03:
Aha! I'm starting to better understand this library and OPC-UA.  By creating a NodeId object with a string identifier and passing it in to add_variable, I control what the id is, and the id can be a string (or even a uuid!)

This has massive positive implications to a lot of other things I'm trying to do, now that the nodes are deterministic.

Thank you so much @oroulet  and @zerox1212

closed 2017-11-03 by ablakey

zerox1212, 2017-11-03:
If you want to use OPC UA I highly recommend you register at OPC Foundation so you can download the spec in PDF.