FreeOpcUa/python-opcua #507: Security with python-OPC UA

GregorPi, 2017-11-03:
Hi,

to finish my bachelor thesis, I want to set up an OPC UA Server on a Raspberry Pi 3. Therefore, I want to use python-opcua to read  in values from some sensors and communicate them over OPC UA.
I already managed to read in the values and put them on a n OPC UA Server. I also can read in these values with an OPC UA Client. 
The only problem left, is the security of the connection. I already found the example-script server-with-description.py but I am not sure how the client gets the certificate.
Further, I want to generate Login-names and passwords to restrict, who can connect to my OPC Server and to generate custom views for the different login-names.

Is this possible to implement with python-opcua?

I would really appreciate help!
Thanks, Gregor

zerox1212, 2017-11-03:
As far as I know the server doesn't send a certificate. The client has to already have a valid one that matches the key on the server.

I haven't tried it, but I think you can create some users on your server here: https://github.com/FreeOpcUa/python-opcua/blob/7542cbbdc618bf5a26f4a1d67d47b70d662d5eef/opcua/server/users.py

Views are not implemented in this library. If you want to implement it and make a merge request that would be great.

oroulet, 2017-11-03:
@zerox1212 This user stuff is really a hack. One should probably use some python user libraries. Should be one somewhere...hopefully...

zerox1212, 2017-11-03:
Can you tell me where passwords are defined?

I have done some user management in the past. One problem I see is that we don't have any place to persist users and passwords. I guess the user code could create them every time server starts for now.

oroulet, 2017-11-03:
There is no password. Just an admin user (Which is disabled by default for
users). If we implement password I would like to do it correctly (and
secure). Maybe using third part libray is a good idea or use system
authentication... I do not know

zerox1212, 2017-11-03:
I have to read the spec on users. It should not be that difficult to have a simple user system with hashed passwords.

oroulet, 2017-11-03:
Maybe. Not sure the spec says anything here. We just need to decide how we want to support users on server side. We probably need a kind of API to register new users and set password. Or use system users? Not sure. But we should probably use the opcua encryption stuff to encrypt passwords so we do not implement a second one! Look at how passwords are encrypted/decrypted

GregorPi, 2017-11-06:
Hi,

thanks for your response. So Views and User Authentification are not (yet) implemented? 

I have a question left about the file 'server-with-encryption.py'. How can I generate my own certificates and keys?

I am very new to this topic, so I am sorry if I am asking stupid questions. ;-)

GregorPi, 2017-11-06:
With the shell-scripts I am able to generate certificates. But i am not sure what they do.

As I understood, they are used to secure the data-transfer between the Client and Server. Is this right?

Thanks, Gregor

zerox1212, 2017-11-06:
I have not tried to use the encryption and it looks like it isn't well documented.

It looks like this library is using a symmetrical key pattern where both client and server have the same private key. The key allows authentication, after that the communications is probably encrypted using only the certificate.

As far as I can tell the security isn't a public/private key style, so keep in mind your security is only secure as long as the private key is safe. If you change the private key on the server, all clients will also need a new key and certificate.

oroulet, 2017-11-06:
As far as I remember it is a public private key solution. The server has a private key. And send public key to clients. but I am really not an expert..

oroulet, 2017-11-06:
@GregorPi the readme tells you how to generate keys

zerox1212, 2017-11-06:
Tests seem to be using the same private key on both client and server.

oroulet, 2017-11-06:
Probably a bad hack

GregorPi, 2017-11-07:
OK. Thanks for all of your comments. It really helped me a lot!

iamjenmackle, 2018-01-06:
Hi @GregorPi, 

I was wondering if you could share your code with me? Or give me a hint on how you "read in values from some sensors and communicate them over OPC UA"? I can't seem to find a method/adapter or anything that does this in this project.

Thanks heaps!

zerox1212, 2018-01-06:
@iamjenmackle Python OPC UA is meant to be used like an SDK.

You can see an example I did demonstrating a basic OPC UA Server which collects data from a device on ModBus.

https://github.com/zerox1212/python-opcua-modbus-driver

GregorPi, 2018-01-07:
Hi,

pathon-opcua is just a library of functions. You can use them in conjunction with functions from other libraries to develop you own software. Unfortunately, I cannot publish my code. But I used a Pixtend Board, which comes with its own library of functions.

I hope, that I could help you.