FreeOpcUa/python-opcua #508: Auto-generated address space code skips some entries

ipa-mdl, 2017-11-06:
The auto-generated code follows the order of the XML tags and therefore references some nodes before they have been registered to address space.

Example from Part10 code:

[ReadyToHalted](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/standard_address_space/standard_address_space_part10.py#L1665)  adds a `HasCause` [reference](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/standard_address_space/standard_address_space_part10.py#L1700) to [Halt](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/standard_address_space/standard_address_space_part10.py#L1703) node, which will get [added later](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/standard_address_space/standard_address_space_part10.py#L1801).
That's why [_add_reference](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/address_space.py#L326) fails silently.

First of all the generated code should check the returned status codes to detect these problems.

I see some ways of fixing the code ordering:
* sort the nodes topologically
* implement a lookup-or-create mechanism to ensure that all node will be created (in the the generated code) before used
* postpone calls to `self.make_refs_code` to be run after all other steps

The topological ordering would also harden to XML importer, which should have the same problem.
I can try to implement this.

ipa-mdl, 2017-11-06:
> Sorting the entire address space will probably be too expensive

Not the entire address space, but the result of [get_node_datas](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/common/xmlparser.py#L132), i.e. in the scope of one XML document.

> Unless you mean to just generate the .py file with a new order.

Yes, for the standard address spaces this will be done during generation of the scripts.

oroulet, 2017-11-06:
@ipa-mdl looks like you made a good analysis. There are two XML parsers one to generate code and one to import directly. Generally XML parsing is the most ugliest part of the stack, so if you manage to improve things it would be great! But postponing ref creation might be the easiest solution... The other methods might get into lock if there is a reference loop...

PR #510 (2017-11-12): Update standard address space
I forgot to generate the standard address space after the the latest patches in #509 

This PR contains the update and a test for references in standard address space.

closes #508
closed 2017-11-12 by oroulet