FreeOpcUa/python-opcua #518: Unpacking Primitive arrays is slow

brubbel, 2017-11-18:
Possibly related to https://github.com/FreeOpcUa/python-opcua/issues/514

Using list comprehension for array unpacking is very slow, for example when reading OPC arrays from history. Profiling narrows it down to the following lines:
https://github.com/FreeOpcUa/python-opcua/blob/9e5a26a97f1079a5db1f55b365411c5e2b5d0317/opcua/ua/ua_binary.py#L203-L207
`return [unpack_uatype(vtype, data) for _ in range(length)]`

l'll create a PR with better performance, where reading 1536 x [8192] arrays from history drops from 60 to <6 seconds.

oroulet, 2017-11-18:
Any help/proposition on this is welcome

brubbel, 2017-11-18:
Thanks, could you also restart the checks in the above PR? Travis seems to fail at random points, which I suspect is due to timeouts while running the tests.

closed 2017-11-18 by brubbel