FreeOpcUa/python-opcua #521: get client connection state

daniel-brosche, 2017-11-20:
I did not find a possibility to check that a client connection has been successfully established like `client.is_connected`.

DaEwe, 2018-02-15:
What worked for me as a workaround is
`client.uaclient._uasocket._thread.isAlive()`

oroulet, 2018-02-15:
I think that with the current implementation, you get a crash is connection is not succesfull. BUT as with any networking system, things may hang for along time depending on timeouts.
If there is a case where the connection is not succefull and you do not get a crash, let me know

DaEwe, 2018-02-15:
Thing is, yes, I get a crash, but how can I react to it programmatically? The exception occurs in another thread..

oroulet, 2018-02-15:
I do not know you application, but usually, people have a kind of ua thread that fetch/push data under a big try/except and restart connection if there is any exception

```

while not self._stop:
    try:
        client.connect()
        node=  client.nodes.server_state    
        while True:
              node.get_value()
              time.sleep(1)
    except:
        logger.exception("something happent, reconnecting in 5s")
        time.sleep(5)
```

DaEwe, 2018-02-15:
I am working with subscriptions.  In your example the exception would be raised from get_value(), right? I can catch that, since it occurred in my own thread. 
How would that work with subscriptions? Just pack my whole application in a try/except-block?

oroulet, 2018-02-15:
for the subscriptions callback this should not change anything, but you need to recreate your subscritions after connecting, since they are lost when disconnecting (ua does allow to reconnect to a subcription, but we do not support that currently, we close them when disconnecting)

oroulet, 2018-02-15:
but if you want to know is connection is lost, you need a kind of polling anyway. This is what clients like uaexpert do. We could have added it into the KeepAlice class, although it is probably more complicated to configure it than just just implementing it

DaEwe, 2018-02-15:
> for the subscriptions callback this should not change anything

Yes it does. The exception is raised within your thread, which is then ended. The rest of my program doesn't notice and so I am not able to handle the situation. Just as in the example below. The prgram will run to it's end and not react to the raised exception.

```import time
from threading import Thread

def some_internal_ua_stuff():
    time.sleep(5)
    raise Exception("Something!")

if __name__ == "__main__":
    try:
        t = Thread(target=some_internal_ua_stuff)
        t.start()
        for i in range(10):
            print("doing sth else")
            time.sleep(1)
        print("done")
    except Exception as e:
        print(" caught exception: " + str(e))
```

oroulet, 2018-02-15:
This is a general issue in multi processing. If a thread dies you need a way to communicate the info to the others, isn'it?

> Much nicer would be either a method of the client as requested by @daniel-brosche, or maybe the possiblity to register a callback.

I know a lot of people seems to want that kind of things. I think it is broken since it is much easier to have my own thread as shown above that monitors the  connection and clean up things if necessary. But is someone make a class for that I am open to merge it (It migth be possible to add that functionnality to the "KeepAlive" class). it should simply be a class started with a polling interval and the possibility to register callbacks to be called if connection is lost, and maybe if connection is remade. It could also have a is_connected attribute if some people want to add one more layer of polling

It has also been propose once to make a ClientForDummies class that automatically reconnects. It iis more work since it needs to keep track of all subscriptions, etc,...

oroulet, 2018-02-15:
and if someone has even more time: The entire client/server API should be ported to the async module to offer an async api. This is more work... an attempte started there: https://github.com/FreeOpcUa/python-opcua/pull/558

OptrixAU, 2020-02-25:
This issue is ancient, but still a legitimate concern. Suggested a small update to include a simple connection test (based on the thread check shown above) to check for dropped connections.

This is particularly useful when you are only subscribing to the OPC - you don't get the opportunity to catch exceptions on connection failure if you're not actively polling. I also considered sending 'hello' requests occasionally, but Hello doesn't appear to be supported by some OPCUA servers.

GLmontanari, 2020-12-02:
Hi!
Any updates about this? I can't find the function "is_connected" in the latest release (0.98.12 from PyPI)

AlexanderSkachkov, 2022-08-10:
I use this in my solution - `opc_ua_client.keepalive.is_alive()`

lorithai, 2025-02-18:
> What worked for me as a workaround is `client.uaclient._uasocket._thread.isAlive()`

You can add the following to ensure it works if you haven't made a connection yet as _uasocket won't be defined 

`
def is_client_connected(client):
    if client.uaclient._uasocket is None:
        return False
    return client.uaclient._uasocket._thread.is_alive()
`