FreeOpcUa/python-opcua #525: object instance with variables

samsa187, 2017-11-23:
i made a custom object with 2 variables and i made an instance of that object where one variable value is being set and the other one is ment for reading value that must set by a client. how can i get the value of the variable to read? 

code:

myobj = objects.add_object(idx, "Device1", mycustomobj_tyoe.nodeid)
mysensor = myobj.get_child(["2:sensor"]) #works
myservo = myobj.get_child(["2:servo"]) # gives the value set in the arg "val" instead of the value set by client 
myservo.set_writable() #doesnt make it writeable i think?

anyone know how to do this?
thanks

zerox1212, 2017-11-23:
Can you give your entire code example because `myservo = myobj.get_child(["2:servo"])` is only going to give you a UA Node object, nothing to do with a value. To get the value which was written to the node by a client you need to call `myservo.get_value()`. If you want to be notified whenever a change is made to `myserver` you need to make a subscription to data change events.

samsa187, 2017-11-24:
in the while loop i trying to get the value of the object node but it gives me the value that is set by the devicetype which is 0. i thought if i make the variable writeable it would give me the value wich is set by the client but it doesnt. 

    # setup our own namespace 1 and 2, not really necessary but should as spec
    uri = "urn:freeopcua:python:server"
    idx = server.register_namespace(uri)
    
    
    
    # get Objects node, this is where we should put our custem nodes
    objects = server.get_objects_node()
    
    # Create a new object type in NS2
    #-------------------------------------------------------------------------------
    types = server.get_node(ua.ObjectIds.BaseObjectType)
    
    mycustomobj_type = types.add_object_type(2, "DeviceType")
    mycustomobj_type.add_variable(2, "USensor", 0.0) # demonstrates instantiate
    mycustomobj_type.add_variable(2, "Servo", 0) # demonstrates instantiate
   
    #-------------------------------------------------------------------------------

    # create instance in NS1
    myobj = objects.add_object(idx, "Device1", mycustomobj_type.nodeid)
    myusensor = myobj.get_child(["2:USensor"])
    myservo = myobj.get_child(["2:Servo"])
    myservo.set_writable()
        
    # setup our sensor
    usensor = v2ultrasonic.ultra()

    #define GPIO pin sensor
    trigg = 18
    ech = 24
    usensor.pin(trigg, ech)

    # define servo GPIO pin
    servo = Servo(17)
    
    # starting!
    server.start()

    print("Running..")
    print ("Namespaceindex: %s" %idx)

    try:
        count = 0
       
        while True:
            
            
            distance = usensor.distance()
            time.sleep(.5)
            
            myusensor.set_value(distance)
            
            servostatus = myservo.get_value()
           
            print("servostatus = %.d" %servostatus)
            
            if servostatus == 1:
                servo.min()
            elif servostatus == 2:
                servo.mid()
            elif servostatus == 3:
                servo.max()
           
    finally:
        # close connection, remove subcsriptions, etc
        GPIO.cleanup()
        server.stop()