FreeOpcUa/python-opcua #526: concurrent.futures._base.TimeoutError

sanlop73, 2017-11-24:
Good morning:

I am having this error at random.
The script sometimes works for hours and suddenly I get this error.
Any solution?

Thanks

Traceback (most recent call last):
  File "opc-client-browse-read-kafka-subscription.py", line 275, in <module>
    read(readpath)
  File "opc-client-browse-read-kafka-subscription.py", line 138, in read
    objects1 = node.get_variables()
  File "/usr/local/lib/python3.5/dist-packages/opcua/common/node.py", line 300, in get_variables
    return self.get_children(refs=ua.ObjectIds.HasComponent, nodeclassmask=ua.NodeClass.Variable)
  File "/usr/local/lib/python3.5/dist-packages/opcua/common/node.py", line 286, in get_children
    return self.get_referenced_nodes(refs, ua.BrowseDirection.Forward, nodeclassmask)
  File "/usr/local/lib/python3.5/dist-packages/opcua/common/node.py", line 360, in get_referenced_nodes
    references = self.get_references(refs, direction, nodeclassmask, includesubtypes)
  File "/usr/local/lib/python3.5/dist-packages/opcua/common/node.py", line 339, in get_references
    results = self.server.browse(params)
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/ua_client.py", line 273, in browse
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

oroulet, 2017-11-24:
Your server does not reply within timeout. Increase timeout

sanlop73, 2017-11-24:
Hi

Are you refering to 
self.session_timeout = 3600000 # 1 hour ??

Could you please provide an example?
Thanks in advance

oroulet, 2017-11-28:
no timeout in client constructor if I remember well. there are many differen timeouts

harua8n, 2017-12-29:
Can you try this:
client = Client("<opc.tcp>", timeout=60*8) # 8 minutes