FreeOpcUa/python-opcua #527: pickle faster than deepcopy

brubbel, 2017-11-26:
Concerning startup speed of opcua on slower hardware, such as raspberry.
https://github.com/FreeOpcUa/python-opcua/blob/51826f9fb44adfe36d13e35324f36d9dce01f758/opcua/server/internal_server.py#L319-L333

Using cPickle (Py2) or pickle (Py3) is faster than deepcopy.
See also: https://wiki.python.org/moin/UsingPickle

Will submit PR as soon as I'm convinced that there are no side-effects.
```
try:
    import cPickle as pickle # python 2.x
except:
    import pickle # python 3.x will automatically load accelerated Pickle version
<...>
    def read(self, params):
        results = self.iserver.attribute_service.read(params)
        if self.external:
            return results
        return pickle.loads(pickle.dumps(results, -1))

    def write(self, params):
        if not self.external:
            # If session is internal we need to store a copy og object, not a reference,
            # otherwise users may change it and we will not generate expected events
            params.NodesToWrite = pickle.loads(pickle.dumps(params.NodesToWrite, -1))
        return self.iserver.attribute_service.write(params, self.user)
```
Left: deepcopy, right: cPickle on Py2.7
[image]

oroulet, 2017-11-26:
Why don't they use pickle in deepcopy implementation then?

zerox1212, 2017-11-26:
If read or write ever runs into an object that can't be serialized with pickle we will crash.

brubbel, 2017-11-26:
Exactly what zerox1212 says.

closed 2017-11-26 by brubbel

zerox1212, 2017-11-26:
Could also do a try catch and fallback to deepcopy.

brubbel, 2017-11-26:
It would be a solution, but the cluttering of the code bothers me.

oroulet, 2017-11-26:
What kind of objects cannot be serialized? We know what we are sending there. It is a variant.

brubbel, 2017-11-26:
This may be useful: https://docs.python.org/3/library/pickle.html#what-can-be-pickled-and-unpickled

brubbel, 2017-11-26:
So, no nested class methods, function code or class variables. But, if it's a variant, could the to/ from_binary be used?