FreeOpcUa/python-opcua #530: Set value range

iamjenmackle, 2017-11-30:
How can you set the value range for a node? Via set_properties? 

An example of a server that simulates mock data would also be greatly appreciated, if any of you have ever done it.

zerox1212, 2017-11-30:
You would have to look in the spec, but I do not think the any of the base types have "range" built in. You would have to make a custom datatype and handle that yourself.

I think Matrikon has a free OPC UA server with sample data. Otherwise you can easily make this python opcua examples.

iamjenmackle, 2017-12-27:
I would like to have a server that simulates two machines with set value ranges. Those machines ought to communicate with each other as well. 

None of the OPC UA examples have set a value range. There is a class called "Range" where you can set a low and a high attribute, also I found something that's called "InstrumentRange" which could also be what I'm looking for. Could you provide an example of how to use it?

zerox1212, 2017-12-27:
You should start with one machine being simulated via one OPC UA server. You will have to build your address space how you like for the machine and write your own code to simulate values. If you post code of what you are trying to do I can help you, but the examples folder should already give you most of the tools to do what you want.

iamjenmackle, 2018-01-06:
Sure thing. As you can tell by the code below, I created an instance of the Range class. With the lowest possible value being 0 and the highest 100. But now, how do I enforce this range on my sensor? The sensor shouldn't be able to have a value outside of this range. Do I add it as a variable? Or do I have to create a reference? 

```python
test_range = ua.Range()        
test_range.Low = 0
test_range.High = 100
```

zerox1212, 2018-01-06:
@iamjenmackle Python OPC UA is meant to be used like an SDK.

You can see an example I did demonstrating a basic OPC UA Server which collects data from a device on ModBus.

https://github.com/zerox1212/python-opcua-modbus-driver

Look at `device.py`. This class should serve as a good example of how to implement things like range, simulation, etc.