FreeOpcUa/python-opcua #531: Cannot historize Custom-Events when 2 Events share the same Property-Name

Nebukadneza, 2017-12-01:
Lets assume we are creating and generating custom events like it is done in `examples/server-events-history.py`. These events have named properties, and are emitted by e.g. an object. It should be possible to have 2 different custom event types with properties of the same name. 

However, when historizing events like this, there seems to be a bug in the SQL historizing backend …

Consider the following incomplete snippet in the context of `examples/server-events-history.py`:
`
    myobj = objects.add_object(idx, "MyObject")
    etype = server.create_custom_event_type(2, 'MyFirstEvent', ua.ObjectIds.BaseEventType,
                                            [('MyNumericProperty', ua.VariantType.Float),
                                             ('MyStringProperty', ua.VariantType.String)])
    etype2 = server.create_custom_event_type(2, 'MyOtherEvent', ua.ObjectIds.BaseEventType,
                                            [('MyMoreOtherProperty', ua.VariantType.Float),
                                             ('MyStringProperty', ua.VariantType.String)])
    myevgen = server.get_event_generator(etype, myobj)
    myevgen.event.MyStringProperty = ua.Variant("hello world")
    myevgen.event.MyNumericProperty = ua.Variant(-456)
    myevgen2 = server.get_event_generator(etype2, myobj)
    myevgen2.event.MyMoreOtherProperty = ua.Variant(1.337)
    myevgen2.event.MyStringProperty = ua.Variant("foo")
`

When executing this and logging the SQL create statements, we get:
`
CREATE TABLE "2_1" (_Id INTEGER PRIMARY KEY NOT NULL, _Timestamp TIMESTAMP, _EventTypeName TEXT, ReceiveTime BLOB, EventId BLOB, Severity BLOB, Time BLOB, MyStringProperty BLOB, Message BLOB, SourceName BLOB, MyStringProperty BLOB, SourceNode BLOB, MyNumericProperty BLOB, EventType BLOB, MyMoreOtherProperty BLOB, LocalTime BLOB)
`
For events of the myobj-Object. Note that there are 2 columns named `MyStringProperty`. This obviously is malformed SQL, hence the table is not created and subsequently not found.

Here, i see 2 bugs:
- There should be logic to distinguish fields-of-eventtypes-of-one-object in the database
- The exception thrown is not a `sqlite3.Error` which is cought and printed, but a `sqlite3.OperationalError` which is silently eaten.

I could probably look into fixing these in the coming week … However if you have some hints or preferences what to do — or are super quick doing it in-between :P — I’d be really glad.

Thanks!
-Dario Ernst

zerox1212, 2017-12-01:
The SQLite backend has some limitations currently. The problem stems from you making two event generators from one object. The current implementation is more or less designed to support one event generator per object (because you basically get one SQL table per object).

To fix this would take some work and redesign. The current code is not a comprehensive implementation.

Nebukadneza, 2017-12-04:
Hi,

thanks for the answer and the clarification!
The first option as suggested by you was what i also had in mind for an intermediate fix for the issue. If that’s a way you could live with (for the moment at least), I’ll gladly come around the corner with some code soon ^_^.

Thanks, and Best Regards,
—Dario

oroulet, 2017-12-04:
Does the dict based history have the same issue?

Nebukadneza, 2018-03-02:
First, sorry for letting this stay idle for so long. I’ve just tried to implement either option, but … kind of failed.

The problem is that `save_event(self, event)` receives a event which has event-props of both event-types. So in the above example, if i trigger `myevgen`, which emits a `etype1`-event-type, which should have only `MyNumericProperty` and `MyStringProperty`, then also `MyOtherProperty` is added to the fields. Now, that function only knows about the Event, it’s `EventType` and `SourceNode` … it cannot know `MyOtherProperty` belongs to a whole different `EventType`.

I’m not sure why this is, or where exactly it happens. I have to admit that deep inside the `InternalSubscription`s my understanding ceased and i lost the code-path from triggering an event into the HistoryManagers subscriptions …. But: On the generating-side, the `Event`-Object didn’t have `MyOtherProperty` yet … which seems more correct to me.

I’d be grateful for a few hints on this ….
Best,
-Dario

zerox1212, 2018-03-02:
The event stuff is very complicated. From what I remember it has a lot of recursion. We would have to wait for @oroulet to tell us why an event sent from the subscription has attributes for ALL event types, instead of just the event that was fired. It could in fact be a bug because I do not think anyone was testing with multiple event generators.

zerox1212, 2018-03-02:
I dug into the code a bit. You should set a break point here https://github.com/FreeOpcUa/python-opcua/blob/e1067baccafcbd0e8c711f6ade0162c792c0e623/opcua/server/internal_subscription.py#L242 and see why `def to_event_fields(self, select_clauses):` is selecting fields for all event types.

I will admit you are getting into the `SelectClause` area of OPC UA and it's rather confusing.