FreeOpcUa/python-opcua #538: Get the value of multiple node at a time

RYLF, 2017-12-27:
I found a get_node method, but I want to get more than one node at a time. Thank you.

zerox1212, 2017-12-27:
Title says you want multiple values. Comment says you want multiple nodes?

At the moment I can only recommend that you write some simple functions in your user code which return lists of the items you want. Otherwise you need to explain more clearly what you are trying to do.

oroulet, 2017-12-27:
If what you want is read value of many nodes at once, then you need to look at code of get_value() and make your own function. Should be very easy

RYLF, 2017-12-28:
hi，this is my code  .it can get a value of one node .  
var = client.get_node("ns=x;s=xxx")
print(var.get_data_value())
I want to get more than one node at a time. Do you have any related API?

For example, to get the ID list of node, and then return to the number of values in the node list at once? Thank

harua8n, 2017-12-29:
@RYLF, try:

`            params = ua.ReadParameters()
            for node_id_str in ['ns=x;s=xxx', 'ns=x;s=xxy']:
                nodeid = ua.NodeId.from_string(node_id_str)
                attr = ua.ReadValueId()
                attr.NodeId = nodeid
                attr.AttributeId = ua.AttributeIds.Value
                params.NodesToRead.append(attr)

            start = datetime.datetime.now()
            result = client.uaclient.read(params)
            print('Read delay : {}'.format((datetime.datetime.now() - start).total_seconds()))
`

RYLF, 2017-12-29:
@harua8n 
Thank you. I'll try.

Mounika-Thota, 2018-02-27:
Hi, I have used the above code i have did small modification to get 30 nodes and node values for every 2 seconds. i kept all 30 nodes in array and i used that array in the for loop
while(1):
            params = ua.ReadParameters()
            for node_id_str in Array:
                nodeid = ua.NodeId.from_string(node_id_str)
                attr = ua.ReadValueId()
                attr.NodeId = nodeid
                attr.Value = nodeid.get_value()
                attr.AttributeId = ua.AttributeIds.Value
                params.NodesToRead.append(attr)
                print params
            time.sleep(2)

and the output is 

ReadParameters(MaxAge:0, TimestampsToReturn:TimestampsToReturn.Source, NodesToRe
ad:[ReadValueId(NodeId:StringNodeId(ns=2;s=173539328), AttributeId:AttributeIds.
Value, IndexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:
StringNodeId(ns=2;s=35129060), AttributeId:AttributeIds.Value, IndexRange:None,
DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=3512
9060_-_1), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:Qualifi
edName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_18), Attribut
eId:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None)), Re
adValueId(NodeId:StringNodeId(ns=2;s=35129060_-_620), AttributeId:AttributeIds.V
alue, IndexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:S
tringNodeId(ns=2;s=35129060_-_110), AttributeId:AttributeIds.Value, IndexRange:N
one, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s
=35129060_-_111), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:
QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_556),
AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:No
ne)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_108), AttributeId:Attrib
uteIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(
NodeId:StringNodeId(ns=2;s=35129060_-_103), AttributeId:AttributeIds.Value, Inde
xRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeI
d(ns=2;s=35129060_-_996), AttributeId:AttributeIds.Value, IndexRange:None, DataE
ncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_
-_1004), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:Qualified
Name(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_109), Attribute
Id:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None)), Rea
dValueId(NodeId:StringNodeId(ns=2;s=35129060_-_3192), AttributeId:AttributeIds.V
alue, IndexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:S
tringNodeId(ns=2;s=35129060_-_734), AttributeId:AttributeIds.Value, IndexRange:N
one, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s
=35129060_-_1191), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding
:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_987),
 AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:N
one)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_106), AttributeId:Attri
buteIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId
(NodeId:StringNodeId(ns=2;s=35129060_-_3272), AttributeId:AttributeIds.Value, In
dexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNod
eId(ns=2;s=35129060_-_3314), AttributeId:AttributeIds.Value, IndexRange:None, Da
taEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=351290
60_-_3315), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:Qualif
iedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_3316), Attri
buteId:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None)),
 ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_3317), AttributeId:AttributeI
ds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None)), ReadValueId(Node
Id:StringNodeId(ns=2;s=35129060_-_2998), AttributeId:AttributeIds.Value, IndexRa
nge:None, DataEncoding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(n
s=2;s=35129060_-_3267), AttributeId:AttributeIds.Value, IndexRange:None, DataEnc
oding:QualifiedName(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_
3268), AttributeId:AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedNa
me(0:None)), ReadValueId(NodeId:StringNodeId(ns=2;s=35129060_-_10), AttributeId:
AttributeIds.Value, IndexRange:None, DataEncoding:QualifiedName(0:None))])

but i couldnt find any particular value for the node. when i see the same nodes in uaexpert we have few values . 
[image]

how can i read the values ? please help

oroulet, 2018-02-27:
@Mounika-Thota this seems to work perfectly this is a low leve ua api returning a low level ReadParameters structure. every structure contains a lot of things inclusing a list of ReadValueId containing the nodeid and and variant which itselv contains the python value...
uexpert also displays all the paramter in your screenshot, not only the value

Mounika-Thota, 2018-02-28:
Thank you for your reply. 
if i want to have only node id and node value for the print statement what changes do i need to do? please help me. In the above output i couldnt see any value

Mounika-Thota, 2018-02-28:
how can i extract only value from all those ReadValueId parameters

zerox1212, 2018-02-28:
You should probably search around the internet for basic python lessons. If you want to print only the value... then just print the value.
`print(my_result.DataValue.Value.Variant.val)`

Mounika-Thota, 2018-03-01:
Iam unable to get the value. Iam getting error while try to print(my_result.DataValue.Value.Variant.val) like 
 'list' object has no attribute 'DataValue'.. I have no idea how to get the simple value.  Please help me.

amisha218, 2019-06-07:
@Mounika-Thota Hey!!….Can you please help me to get the values of multiple tags/node at a time.

how did you resolve this issue??

Keul125, 2019-07-17:
Thanks to @RYLF :
```
        tableau = []
        tableau.append('ns=3;s=Link_S7_1.db1280.12,b')
        tableau.append('ns=3;s=Link_S7_1.db1280.16,x5')
        params = ua.ReadParameters()
        for node_id_str in tableau:
            nodeid = ua.NodeId.from_string(node_id_str)
            attr = ua.ReadValueId()
            attr.NodeId = nodeid
            attr.AttributeId = ua.AttributeIds.Value
            params.NodesToRead.append(attr)

        start = datetime.datetime.now()
        result = client.uaclient.read(params)
        print('Read delay : {}'.format((datetime.datetime.now() - start).total_seconds()))
        for rep in result:
            # print(rep)
            print(rep.Value.Value)
```

Syntax for S7 data type:
,xN => Bit number N (ex: db1280.84,x0 )
,b => byte
,c => char
,w => word
,i => int
,d => unsigned double word
,di => signed double word
,r => real/float

vidyarao123, 2021-01-29:
Hi , To the same question, with the below code, I am able to write for 1 tag at a time. Is it possible to write for multiple tags at the same time ?

name = "PYTHON_OPCUA_SERVER"
addspace = server.register_namespace(name)
node = server.get_objects_node()
Param = node.add_object(addspace, "Parameters")
Temp = Param.add_variable(addspace, "Temperature", 0)
Temp.set_value(2)

AndreasHeine, 2021-01-29:
sure! client.set_values()

vidyarao123, 2021-01-29:
But I am trying from the server side

AndreasHeine, 2021-01-29:
unfortunately we only provide a highlevelmethod on clientside because its part of the spec.!

this is actually only a conviniencefunction (at leat in my eyes)... you could try to make a localhost client connection to the server and use the request but then it must be writeable from clientside and is i would assume not more performant then looping over a list and write each item. on serverside there should not be an major delay!