FreeOpcUa/python-opcua #544: ExtraPaddingSize not implemented

Infinity95, 2018-01-14:
The current implementation does not seem to support key sizes > 2048 bit.
For keys > 2048 bit an extra padding byte is needed because the padding can be longer than 255 byte.
Spec part 6 v1.03 Table 30 or Spec part 6 v1.04 Table 45

When creating padding this needs to be taken into account [here](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/crypto/security_policies.py#L126) and when removing padding [here](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/crypto/security_policies.py#L168).

oroulet, 2018-01-15:
Ok. If you know that kind of things, it would be great if you send a PR

Infinity95, 2018-01-15:
I will make a PR. I was waiting for someone to confirm this before i start changing stuff though.

oroulet, 2018-01-15:
I wrote most of python-opcua code but I know nothing about encryption and code has been mainly written by @alkor. So he is probably the guy able to comment on this

zerox1212, 2018-01-15:
If you know cryptography, please also check the public/private key situation. I think there is some problems there.

Infinity95, 2018-01-15:
Can you be a bit more specific than some problems, or are you just unsure if all is well?

zerox1212, 2018-01-15:
Last time I looked there was no public key. Both the server and and the client use the same private key for encryption.

oroulet, 2018-01-15:
I kind of remember using same certificate for testing. But there should be no issue using 2 different certificates

oroulet, 2018-01-15:
@Infinity95 interesting software you work on....

Infinity95, 2018-01-15:
The Security policies have the following parameters in their init method:

```
class SecurityPolicyBasic256(SecurityPolicy):
    def __init__(self, server_cert, client_cert, client_pk, mode):
        ....
```
The naming here is a bit misleading. IMHO it would be better to name it like this:

```
class SecurityPolicyBasic256(SecurityPolicy):
    def __init__(self, remote_cert, local_cert, local_pk, mode):
        ....
```
So when communicating with a remote (either as server or client, it doesn't matter) we need the remote certificate and our local certificate and the matching private key. The usage seems fine from the quick look i had.

oroulet, 2018-01-15:
@Infinity95 i agree

zerox1212, 2018-01-15:
Since you seem knowledgeable about security. Can I ask why we do not need a pair of keys (public/private)? Instead we have a single private key that is shared between all clients and servers.

Infinity95, 2018-01-15:
We have a pair of keys. The public key is contained in the certificates. In the asymmetric encryption case we use the public key in the certificate to encrypt the message (everybody can encrypt messages this way) and then the recipient can decrypt the message with the private key that belongs to the certificate. Since the private key is only known (or at least should be) to the recipient, he is the only one who can decrypt the message.

One can of course use the same private key on both sides but that defeats the purpose of asymmetric encryption and one could just as well use symmetric encryption.

I haven't looked at the examples yet, but the client and server can be loaded with a different private key and corresponding public key.

zerox1212, 2018-01-15:
Thanks for the detailed answer.

> One can of course use the same private key on both sides but that defeats the purpose of asymmetric encryption and one could just as well use symmetric encryption.

So it could be that the implementation allows asymmetric encryption, but the example and the tests are using symmetric encryption. I would really appreciate if you could show an example using asymmetric encryption.

oroulet, 2018-01-15:
Yes tests probably use one certificate to simplify things