FreeOpcUa/python-opcua #547: c++ keep getting response timeouts after update

codersmurf, 2018-01-18:
Not sure what the version was before, probably at least a year old, but updated this week and can't get the app to be stable anymore.  I couldn't figure out what was wrong, the exceptions unwound my stack so I took out the throw that brought me the exception and found that it was getting the browse name that kept having the problem.  Come to find out the browse actually shot off a request to the server, so I took all the GetBrowseName calls to once per node and kept that var around.  Same problem, just not as often now.  Also getting warnings about unknown subscriptions, but I couldn't find a way to see if a subscription came back good, I assume an exception would take me down in that event.  So I'm kind of in a holding pattern and not sure where to go, should I be checking if each node is valid when traversing children, thought a failure there would throw, should I slow down the app more with sleeps, any advise would be much appreciated. Thx

zerox1212, 2018-01-18:
You did not change your code and only installed the latest library? Are you using python opcua from pip or from the git master?

codersmurf, 2018-01-18:
c++, library from master, so the code is pretty close to original, usually don't need to touch it, but they wanted a new feature.  So I added some more subscriptions, I disabled those features after running into the problem, but for the most part it's the same.  Also, I saw a post where after looping through GetChildren() I should be sleeping before using the node, so I moved my sleep up to the top instead of the bottom of the being used, same problems, but I've been running in debug now for like 10 minutes after that change, a new record after all the changes.  Also, in debug mode the unknown subscriptions go away, but I'd like to have logs and not be flooded, so it's definitely a timing thing, just not sure were best place is for them, as it seems to be in the freeopcua lib threads, but I would think I could slow things down on my end to increase the stability.  Also, as a side note, I've always had stability issues, but just added a daemon that would restart it when it noticed it not running, didn't happen all that often but enough to warrant the daemon. Thx

codersmurf, 2018-01-19:
Hahaha, never mind about the running in debug, noticed it died, looked at logs I was tailing and it died shortly after starting, the debug output just took forever to write it all out.