FreeOpcUa/python-opcua #548: URI being registered before user is able to change it.

JuanoD, 2018-01-18:
You are supposed to use the method server.set_application_uri("blablabla") to set ehe application uri, but in the __init__ method, it registers the default uri before you can change it, causing an extra uri to appear when you do server.get_namespace_array()

```python
        def __init__(self, shelffile=None, iserver=None):
        self.logger = logging.getLogger(__name__)
        self.endpoint = urlparse("opc.tcp://0.0.0.0:4840/freeopcua/server/")
        self.application_uri = "urn:freeopcua:python:server" # This one here will always be registered
        self.product_uri = "urn:freeopcua.github.no:python:server"
        self.name = "FreeOpcUa Python Server"
        self.application_type = ua.ApplicationType.ClientAndServer
        self.default_timeout = 3600000
        if iserver is not None:
            self.iserver = iserver
        else:
            self.iserver = InternalServer(shelffile)
        self.bserver = None
        self._discovery_clients = {}
        self._discovery_period = 60
        self.certificate = None
        self.private_key = None
        self._policies = []
        self.nodes = Shortcuts(self.iserver.isession)

        # setup some expected values
        self.register_namespace(self.application_uri) # Because this is here, inside the __init__
        sa_node = self.get_node(ua.NodeId(ua.ObjectIds.Server_ServerArray))
        sa_node.set_value([self.application_uri])
```
Result of the server.get_namespace_array()
```python
    ['http://opcfoundation.org/UA/', 'urn:freeopcua:python:server', 'http://opcfoundation.org/UA/DI/', 'http://fdi-cooperation.com/OPCUA/FDI5/', 'my:uri']
```
I think it would be better to enforce the user to register the uri after creating the server.
My main trouble here is I have required uris in my xml model, so I need to register them before my application uri.

I'll locally edit the library so I can have the uri as an argument on server creation, then I'll register my first required uri, and after I finish the xml importing, I'll change the uri to the one I want my server to have.

```python
    def __init__(self, desired_uri="urn:freeopcua:python:server", shelffile=None, iserver=None):
        self.logger = logging.getLogger(__name__)
        self.endpoint = urlparse("opc.tcp://0.0.0.0:4840/freeopcua/server/")
        self.application_uri = desired_uri
```
And then
```python
    server = Server(desired_uri="http://opcfoundation.org/UA/DI/")

    print(str(datetime.now()), "-> 0/3")
    server.import_xml("NodeSets/Opc.Ua.Di.NodeSet2.xml") # uri="http://opcfoundation.org/UA/DI/"
    print(str(datetime.now()), "-> 1/3")
    server.import_xml("NodeSets/Opc.Ua.Fdi5.NodeSet2.xml") # uri="http://fdi-cooperation.com/OPCUA/FDI5/"
    print(str(datetime.now()), "-> 2/3")
    server.import_xml("NodeSets/Opc.Ua.Model.NodeSet2.xml") # uri ="my:uri"
    print(str(datetime.now()), "-> 3/3")

    server.set_application_uri("my:uri") # Finally, set the server uri to my:uri
```
Which gives me as result of print(server.get_namespace_array()) this:
```python
    ['http://opcfoundation.org/UA/', 'http://opcfoundation.org/UA/DI/', 'http://fdi-cooperation.com/OPCUA/FDI5/', 'my:uri']
```

I know is a dirty workaround but I have a deadline, so no time to figure a better solution.

PD: Great job, guys, keep it going.

oroulet, 2018-01-18:
Thanks for the report. Maybe this should be moved to _setup_server_nodes().... Not sure there was a reason for having this here... No need for a setter either since we have a property..

JuanoD, 2018-01-22:
#550

closed 2018-01-22 by JuanoD

oroulet, 2018-01-31:
Looks like that needed one more fix: https://github.com/FreeOpcUa/python-opcua/pull/553 otherwise the order get wrong. appliation uri should be index 1

JuanoD, 2018-01-31:
I always have my index at 3, because I use OPC DI and FDI nodesets. I would rather to have application uri at last index. Is it really necessary for application uri to be index 1?