FreeOpcUa/python-opcua #549: BadAttributeIdInvalid

JuanoD, 2018-01-22:
Some nodes appear with this error and I don't understand why.
[image]
For example, I have this nodes. The childs of Identification are linked via an "Organizes" reference, while the ones in ParameterSet are linked via an "HasProperty" reference. 
[image]
[image]

It also happens with variables, and even some objects under a HasComponent reference
[image]

oroulet, 2018-01-22:
It just means that uaexpert expects som attributes that a node does not
have. usually, this is not an issue. But is this is part of a model that
you have imported and is missing, then maybe we have an issue in parser.

JuanoD, 2018-01-22:
What is weird is that the nodes are displaying correctly under ParameterSet, but not under the Servicing and Identification nodes. They are the same nodes, they just are being referenced differently. Maybe I'm making a mistake trying to Organize properties, but the variables should work with that. If you want, I can upload my files so you can see.

oroulet, 2018-01-27:
If you have a file i would like to have a look. Looks like i eill do some ua related work the next weeks

JuanoD, 2018-01-27:
https://github.com/JuanoD/I4o-Project
Here, I'm really new to this library, so I'm still figuring some things.
Can I ask you some stuff about how some things work?

JuanoD, 2018-01-28:
Basically, right now I need to accomplish two things.

The first one is to create a simple user to be able to write from client, it is not necessary to use password encryption or a database, it is for educational purposes, so everyone can look at how it works and make their new user management based on that basis.

The second one is to add a server to an existing server. What I mean is, I have a running server on a device and I want a second server running on another device on same network to add its adress space to the first one.

oroulet, 2018-01-28:
* users
currently there are only two users defined in users.py. If someone want to create a more advanced systems with internal users then he should look at uaprocessor.py to restrict session creation for existing users and maybe at session object i internal_server.py (could be split out of that file btw)

* adding servers
What do want to exactly? forward request from one server to another on specific namespace?. What part of spec is that? We need then to add a check for namespace somewhere and make a network call...That may require playing with async code again...

JuanoD, 2018-01-28:
- I'll have a look at how users are handled, if I have some questions, I'll ask you. Thank you very much.

- I think I misread the spec about adding servers. I haven't read the entire spec, but in part 3, 4.4.2, it says:
    > A Node and its Properties shall always reside in the same Server.

    That made me think some servers can have references to another already existing servers.

JuanoD, 2018-01-28:
- I'll have a look at how users are handled, if I have some questions, I'll ask you. Thank you very much.

- I think I misread the spec about adding servers. I haven't read the entire spec, but in part 3, 4.4.2, it says:
    > 4.4.2 Properties
    >...
    >A Node and its Properties shall always reside in the same Server.

    That made me think some servers can have references to another already existing servers.

harua8n, 2018-01-29:
RE: That made me think some servers can have references to another already existing servers.
Yes, I have read this as well in "OPC Unified Architecture" by Mahnke, Wolfgang, et al. I have yet seen this in spec though.

oroulet, 2018-01-29:
yes. there is something about nodes belonging to a namespace in another server. Never looked at it closely...

JuanoD, 2018-02-08:
I just noticed I can't reverse browse nodes with this issue. I mean, when I try to get parent, it says the node has no parent.