FreeOpcUa/python-opcua #552: Minimum Subscription period allowed

sanlop73, 2018-01-25:
Good afternoon:

I m trying to catch item changes in milliseconds with subscriptions.
Signal is changing in the PLC about 100 ms using a Boolean values.
What is the minimum value to set  up the subscription?

Example: sub = client.create_subscription(100, handler)
Problem i am loosing some status changes.

Thanks a lot for your comments

zerox1212, 2018-01-25:
I had the same question here https://github.com/FreeOpcUa/python-opcua/issues/288 , but this issue is more specific so we should pick it up here.

My experience from the server is that I can't tell if the code is actually obeying the update rate. We might have to really dig into the code to find if passing `100ms` has any effect.

sanlop73, 2018-01-28:
Good afternoon:
I have checked with wireshark and the parameter PublishInterval is required and acknowledge.
See screenshots.
[image]
[image]
[image]

Any ideas?

harua8n, 2018-01-29:
I have seen servers that does not obey the RevisedPublishingInterval, this is specially the case when it is under load. I am not sure what the specs tells us but at this point OPC UA may not be designed for real-time applications.

oroulet, 2018-01-29:
If callback is not called at all then there are  probably some issues with server overload. But anyway data change protocol is wrong choice if you do not want to miss changes. You need opcua events.... If your server supports it.....

sanlop73, 2018-01-29:
Good morning:

Looks like we need to modify monitored items.

Got it from ProsysOPC
https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/

**SamplingInterval and QueueSize**
_If you wish to record samples at a faster rate than PublishingInterval, you should use the SamplingInterval of the monitored items. You will also need to reserve a longer QueueSize, to be able to keep all the samples in the server. The server can send all the samples in the queue in one notification – which is sent every PublishingInterval._

So I think to achive this , we need to Modify Monitored Items (Sample Time parameter)

    def modify_monitored_item(self, handle, new_samp_time, new_queuesize=0, mod_filter_val=-1):
        """
        Modify a monitored item.
        :param handle: Handle returned when originally subscribing
        :param new_samp_time: New wanted sample time
        :param new_queuesize: New wanted queuesize, default is 0
        :param mod_filter_val: New deadband filter value
        :return: Return a Modify Monitored Item Result
        """
        for monitored_item_index in self._monitoreditems_map:
            if self._monitoreditems_map[monitored_item_index].server_handle == handle:
                item_to_change = self._monitoreditems_map[monitored_item_index]
                break
        if mod_filter_val is None:
            mod_filter = None
        elif mod_filter_val < 0:
            mod_filter = item_to_change.mfilter
        else:
            mod_filter = ua.DataChangeFilter()
            mod_filter.Trigger = ua.DataChangeTrigger(1)  # send notification when status or value change
            mod_filter.DeadbandType = 1
            mod_filter.DeadbandValue = mod_filter_val  # absolute float value or from 0 to 100 for percentage deadband
        modif_item = ua.MonitoredItemModifyRequest()
        modif_item.MonitoredItemId = handle
        modif_item.RequestedParameters = self._modify_monitored_item_request(new_queuesize, new_samp_time,
                                                                             mod_filter, item_to_change.client_handle)
        params = ua.ModifyMonitoredItemsParameters()
        params.SubscriptionId = self.subscription_id
        params.ItemsToModify.append(modif_item)
        results = self.server.modify_monitored_items(params)
        item_to_change.mfilter = results[0].FilterResult
        return results

Unfortunatelly i have not examples available.
Any help here?

Thanks

abcarcol, 2019-09-20:
> If callback is not called at all then there are probably some issues with server overload. But anyway data change protocol is wrong choice if you do not want to miss changes. You need opcua events.... If your server supports it.....

Could you please tell me the difference between OPC UA data change subscriptions and OPC UA events?

Thank you so much in advance!