FreeOpcUa/python-opcua #557: References to address space in other server.

JuanoD, 2018-02-08:
As I previously pointed in #549, there must be a way for referencing nodes on other servers.

This is defined in IEC 62541 (OPC UA Specification) part 3

> ### 4.3.4 References
>...
>
> The TargetNode of a Reference may be in the same AddressSpace or in the AddressSpace of
another OPC UA Server. TargetNodes located in other Servers are identified in OPC UA
Services using a combination of the remote Server name and the identifier assigned to the Node
by the remote Server.

I just re-submitted it in its own issue so developers can identify it as a feature request or improvement. I'll try to have a look on how I can make it work, but I don't promise anything.

Thanks for your attention, and great job, guys.

zerox1212, 2018-02-08:
Have you found an OPC UA server that can actually do this?

JuanoD, 2018-02-09:
To be honest, I haven't tried many servers. I tried to use open62541 but I was unable to import my namespace. This was always my first option.

oroulet, 2018-02-09:
It sounds like s very simple thing to implement anyway.  Maybe nothing to do. Just reply to client that node is from another server. It does not seem to rewrite any synchronization functionality