FreeOpcUa/python-opcua #559: Input Client.set_security()

maxveenman, 2018-02-16:
I am trying to connect with a server which has no option to communicate with any other policy method than 'Basic128Rsa15' or 'Basic256'. Therefore, I have to set this with the function 'Client.set_security()'.

Client.set_security asks for the following arguments:
_self_, policy, certificate_path, private_key_path, server_certificate_path=None, mode=ua.MessageSecurityMode.SignAndEncrypt

I managed to find all these arguments except the 'policy'. This seems to be a class of some kind, because in Client.set_security() I find 'policy.URL' and 'policy(server_cert, cert, pk, mode)'.

I cannot find a function or class that is initialized with the 4 arguments 'server_cert, cert, pk, mode'.

What exactly do I need to input to Client.set_security() **policy** argument?

zerox1212, 2018-02-16:
Did you try looking in the tests? Perhaps there is an example there, but I'm not sure.

maxveenman, 2018-02-17:
No, I only looked at the example files. In the test files I indeed found a working example. Thank you very much!

closed 2018-02-17 by maxveenman