FreeOpcUa/python-opcua #560: Getting datas from Terminal

Escagod, 2018-02-25:
Hello,
I am currently working on a project where OPC Clients must be able to get datas from the Terminal through the OPC Server. I have a lot of sensors, which print out their datas continuously on the Terminal and other OPC Clients must be able to get the datas as soon as possible after the print. Can you please explain me which of the example codes I have to use and how I have to configure it for my project? I would really appreciate your help very much.

uprightcarrion, 2018-02-26:
I've used this to gather data from controllers in an industrial setting.  To do what your looking for I would look at the [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py](url).  I'm far from a master of these but essentially you subscribe to a variable at a desired interval.

uprightcarrion, 2018-02-26:
That link doesn't seem to work, not sure why....look at the path though and you'll find it.

zerox1212, 2018-02-26:
You need to design an address space for your project. After that you need to start an OPC UA server, setup the address space, and finally populate the nodes of your address space with data (for example from your sensors). Then OPC Clients will be able to connect to your server and subscribe to the nodes that hold your data.

uprightcarrion, 2018-02-26:
I may have misunderstood but I think Escagod has a terminal which offers a server already in which case only the client is necessary.  If the sensors or terminal don't provide a server then for sure you need to setup your own server.

Escagod, 2018-02-26:
Thanks for replying. 
@uprightcarrion I have installed everything which is necessary to start a OPC server, but I don't have any basic knowledges how I have to setup my server for my project. That's why I have some difficulties to create a code, which will work for my project. Let me explain you everything again.
I have a Raspberry Pi and some sensors, which I have programmed with Python and they are printing out their datas after a certain intervall. For example I have a temperature sensor, which prints out the temperature of my room every two seconds on the Terminal of Raspbian. Now I have to create a OPC Server where OPC Clients can get the datas of my sensor, which are printed out on the Terminal.
@zerox1212 Thank you for explaining, but can you maybe give me a example code? Because I have really no idea how to do it.

oroulet, 2018-02-26:
@Escagod it sounds like you need to learn som basic concepts first. Make sure that you need OPC UA and not OPC DA,  then get some experience with python programming then look a the examples in examples directory.

zerox1212, 2018-02-26:
Go to the examples folder and start with `server-minimal.py`. Make your own attempt to get it working and post your code and if we have time we can help you.

Escagod, 2018-03-03:
Hello guys,

I have used a code from the examples and tried to get it working. But there are still some issues, which have to be solved. First of all I want to share you my code,

`import sys
sys.path.insert(0, "..")
import time
import datetime
import subprocess
import logging

from opcua import ua, uamethod, Server

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://10.0.0.112:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()

    # populating our address space
    sensordata = objects.add_object(idx, "Sensordata")
    tsl2591 = sensordata.add_variable(idx, "tsl2591", 6.7)
    tsl2591.set_writable()    # Set MyVariable to be writable by clients

    # starting!
    server.start()
    
    try:
        count = 0
        
        while True:
            #run luminosity sensor tsl2591
            tsl2591 = subprocess.call(" python tsl2591.py 1", shell=True)
    
    finally:
        #close connection, remove subcsriptions, etc
        server.stop()`

10.0.0.112 is the IP address of my raspberry pi, which I use for my server. A Matlab client was trying to connect to my server through the public IP address of my router, but it was not working. The Error "A low level communication error occured. Ensure that server host is reachable" always occured. I have already opened the port 4840 for my router, but it just won't work. Every help would be appreciated.

oroulet, 2018-03-03:
That is a network issue. Nothing to do with this library...