FreeOpcUa/python-opcua #562: Error: With latest version 0.95

kapsl, 2018-02-27:
Hi,
we used the opc ua client without problems the last months. But we tried to update from 0.90.4 to 0.95 and suddenly we get following error, which causes the client to crash...

opcua.uaprotocol - WARNING - Received an error: MessageAbort(error:StatusCode(Good), reason:Bad_DecodingError (code=0x80070000, description="Unexpected NodeId Encoding Byte -127"))
2018-02-27 11:55:09,755 - opcua.client.ua_client.Socket - WARNING - Received an error: MessageAbort(error:StatusCode(Good), reason:Bad_DecodingError (code=0x80070000, description="Unexpected NodeId Encoding Byte -127"))

lg Manu

oroulet, 2018-02-27:
There have some reither fundamental changes in the releases, that may explain. We need more information to debug that. Ideally fin exactly what node gives that error message, and what is special with it. If you do no manage that then  you need to record the failing session with wireshark and paste it here

kapsl, 2018-03-05:
So basically, this is a very normal connection to a ProSys OpcUa Server. 
The subscription is made in a way that we have a queue, and do not transfer every single value.

oroulet, 2018-03-07:
I tested connection to a prosys opcua server this morning, found a small bug when authenticating using certificates, But otherwise things work fine. We need more info as written last week