FreeOpcUa/python-opcua #563: Write data to OPC UA server

alfalog, 2018-03-07:
Hi,
I have written a python program that generates some calculations as output on Terminal. Now, I want to write this data to OPC UA server (prosys , freeopcua server or any other OPC UA server). How can I create a client-server communication and write all the data from terminal to OPC server?  Please help!

zerox1212, 2018-03-07:
The examples folder has everything you need.

alfalog, 2018-03-07:
@zerox1212:- I already tried the examples from the folder. But they are not working. The server-example.py is having errors with asyncio.subscription. The client-example.py is unable to communicate to server. Also, client and server -minimal.py are not working together for connection. What do you suggest?

zerox1212, 2018-03-07:
Did you install Python OPC UA with pip? Or are you using the source code?

I would first start a server and make sure it's working by using the free version of UA Expert. https://www.unified-automation.com/downloads/opc-ua-clients.html

After you have a known working server you can run the python client.

alfalog, 2018-03-07:
I installed opcua using pip. As said in the code documentation.
Is there any problem with pip installation?
And Thanks for your suggestions!

oroulet, 2018-03-07:
Opcua from pip works fine. i saw someone installing and using it today. Something is wrong on your PC. Check firewall for example and paste errors her

alfalog, 2018-03-07:
So, these are the errors I am getting when I run the programs:
from the examples folder:
python server-minimal.py

No handlers could be found for logger "opcua.server.binary_server_asyncio"

And on the other terminal I executed 'client-minimal.py' I got following:

('Objects node is: ', Node(TwoByteNodeId(i=84)))
('Children of root are: ', [Node(NumericNodeId(i=85)), Node(NumericNodeId(i=86)), Node(NumericNodeId(i=87))])
Traceback (most recent call last):
  File "client-minimal.py", line 33, in <module>
    myvar = root.get_child(["0:Objects", "1:MyObject", "1:MyVariable"])
  File "../opcua/common/node.py", line 459, in get_child
    result.StatusCode.check()
  File "../opcua/ua/uatypes.py", line 235, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNoMatch: The requested operation has no match to return.(BadNoMatch)

oroulet, 2018-03-07:
Your installation works perfectly. The last error is just a wrong namespace. This has been fixed in master

alfalog, 2018-03-07:
Thanks a lot..!

closed 2018-03-07 by alfalog