FreeOpcUa/python-opcua #564: Server Terminal to Client

alfalog, 2018-03-08:
I am working on a project that executes a program on the server side. And prints the output on the terminal of OPCUA server. But it is not printing anything on the client side.
I am getting the following error on the client side:

------------------------------------------------------------
Traceback (most recent call last):
  File "client.py", line 34, in <module>
    print(resulting_data)
  File "client.py", line 15, in __exit__
    self.client.disconnect()
  File "build/bdist.linux-x86_64/egg/opcua/client/client.py", line 268, in disconnect
  File "build/bdist.linux-x86_64/egg/opcua/client/client.py", line 486, in close_session
  File "build/bdist.linux-x86_64/egg/opcua/client/ua_client.py", line 261, in close_session
  File "build/bdist.linux-x86_64/egg/opcua/client/ua_client.py", line 77, in send_request
  File "/home/vd/.local/lib/python2.7/site-packages/concurrent/futures/_base.py", line 464, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError
-----------------------------------------------------------------

What changes should I make, to solve this issue?

And is there any code or any reference link that can help me for transferring Server terminal data to Client?

My work is similar ad mentioned in issue #560 .

zerox1212, 2018-03-08:
It looks like a network issue. You are getting a timeout.  Are you running the server and client on the same PC? Do you have firewall disabled? Can you ping your server from the client PC?

alfalog, 2018-03-08:
I am running client server on the same pc. And my firewall is also disabled

closed 2018-03-08 by alfalog