FreeOpcUa/python-opcua #568: Do we have a turorial about how to use opcua tools?

leonwanghui, 2018-03-27:
Hi all, recently I am learning to install and run this project. When I ran `pip install opcua` command, there was a lot of command line tools installed in my environment (which is great), but I didn't find any related docs (even code) to use it. So do I miss something or is the docs somewhere else? Thanks!

oroulet, 2018-03-27:
Good point.... Where should we document them? Run 'uals --help' for example

leonwanghui, 2018-03-27:
@oroulet Yes I got it, thanks. Besides, I also found the code in `/usr/local/lib/python2.7/dist-packages/opcua/tools.py` about how it actually works.

But IMO we need to add some documents for guiding non-python developers to use it, for example we could show some examples about how to call these command line tools, like what we are doing in `examples/client-examples.py`.

Any thought?

Keul125, 2019-07-17:
Example for reading data:
Press Win+R, and type "cmd" to onen a terminal (executing the .exe will close immediately) then type
"C:\Users\YOURACCOUNT\AppData\Roaming\Python\Python37\Scripts\uaread.exe" -u opc.tcp://127.0.0.1:55101/Siemens/SimaticNet/OpcUaServer/S7 -n ns=3;s=Link_S7_1.db1280,b

to get the graphical UI, type "pip install PyQt5" and
python "C:\Users\YOURACCOUNT\AppData\Roaming\Python\Python37\Scripts\opcua-client-script.py"

Syntax for S7 data type:
,xN => Bit number N (ex: db1280.84,x0 )
,b => byte
,c => char
,w => word
,i => int
,d => unsigned double word
,di => signed double word
,r => real/float