FreeOpcUa/python-opcua #571: Subscription Keepalive

sanlop73, 2018-04-01:
Hi:

I need to check if the subscription is alive, read some threads here but i have no clear conclusion.
Could you please give me a hint , how to implement this?
Do you have an example?

Thanks in advance

oroulet, 2018-04-01:
What do you want to do exactly? Why? The subscription itself sends keepalive messages but they are not really exposed. You could implement your own stuff using uaclient class

sanlop73, 2018-04-01:
Thanks for your reply.
I have a subscription which their items changes aproximatelly every 8 hours, so the problem is can check if the subscription is still alive. That why i m asking for the keepalive.

Do you have an example how to expose?

Thanks again.

DionisisBozikis, 2020-04-27:
hello i am interested in a keepalive example, also. 
thanks