FreeOpcUa/python-opcua #572: Updating address space to 1.04

cirp-usf, 2018-04-05:
Hi,

In the project I'm working we need the 1.04 version of the standard address space. So I regenerated the files in [opcua/server/standard_address_space/](../tree/master/opcua/server/standard_address_space/) using [schemas/generate_address_space.py](../tree/master/schemas/generate_address_space.py) and the new NodeSet files. I needed to  manually reorder one of the generated files.

The generated files allow me to import the NodeSet of the [specification] I’m working with and I’ve been using them ever since.

I would like to contribute the generated files back. However now I realize other files may need to be changed as well, for instance [opcua/ua/object_ids.py](../tree/master/opcua/ua/object_ids.py). 
In order to make a PR, do I have to update other files as well? If so, which files must be updated and how?

Best regards,
Ubiratan

[specification]: http://www.euromap.org/en/euromap77

oroulet, 2018-04-05:
There is a file called download address space in schema. Then when calling generate_protocol.py, all these files should be updated

oroulet, 2018-04-06:
Make a pr anyway so we can see and discuss issues. Concerning testing. You can run python3 /tests/tests.py

oroulet, 2018-04-06:
It is automatically run when you create a pr btw...

cirp-usf, 2018-04-06:
PR made. Didn't pass the tests. I'll look at it.

cirp-usf, 2018-04-07:
I saw only 4 failures at the output of the tests. They are all related to the _AttributeWriteMask_ standard type. In version 1.04, it is defined in part 3 (8.20, p. 73) as a subtype of UInt32. I could not find it in 1.03, but in `schemas/Opc.Ua.NodeSet2.Part4.xml` of 1.03 this type was defined as a subtype of _Enumeration_. Two test files (`tests_common.py`, line 927 and `tests_xml.py`, line 305) explicitly attribute the Int32 data type to _AttributeWriteMask_. Those are the tests that fail.

Would it be OK to change those lines to use UInt32? With that change those tests pass, but I think that this would make the test in `tests_xml.py` pointless. This test is called `test_xml_enumvalues`, which _AttributeWriteMask_ no longer is.

oroulet, 2018-04-07:
OK I had a loook and you can change type. This does not change anything in binary protocol