FreeOpcUa/python-opcua #573: add_object() from user defined type does not inherit subnodes.

MikelAyani, 2018-04-06:
Hi,
I have seen after the last update today (using pip3) that when instancing (add_object()) user made ObjectType and VariableType, their children nodes are not inherited. I guess it is a bug in the last version because it worked before....
Thanks!

oroulet, 2018-04-06:
Can you dobbel check that it is not a user error. I used it yesterday from master and it worked fine...

MikelAyani, 2018-04-06:
Sorry, it works with ObjectTypes but not with VariableTypes which have Variable children.

oroulet, 2018-04-06:
You should open a bug report for that. We just allowed it and we should also instantiste children of variable... Even better make a pr

MikelAyani, 2018-04-06:
Ok, thanks! But how can I open a bug report?

cirp-usf, 2018-04-06:
I've encountered this problem before. As a work around I've been using add_object() to add variable types. I just pass a VariableType to add_object() and the children get instantiated.

MikelAyani, 2018-04-06:
Yes, this is what I have been doing, but since last update add_object with a VariableType does not inherit children...