FreeOpcUa/python-opcua #578: An error about run-test.py

WangAllen, 2018-04-17:
Thanks for all
After doing ./run-test.py, there are 4 ERROR under here, how to deal?
======================================================================
ERROR: test_uadiscover (tests_cmd_lines.TestCmdLines)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/python/Desktop/python-opcua-master/tests/tests_cmd_lines.py", line 46, in test_uadiscover
    s = subprocess.check_output(["python", "tools/uadiscover", "--url", self.srv_url])
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python', 'tools/uadiscover', '--url', 'opc.tcp://127.0.0.1:48530']' returned non-zero exit status 1

======================================================================
ERROR: test_uals (tests_cmd_lines.TestCmdLines)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/python/Desktop/python-opcua-master/tests/tests_cmd_lines.py", line 28, in test_uals
    s = subprocess.check_output(["python", "tools/uals", "--url", self.srv_url])
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python', 'tools/uals', '--url', 'opc.tcp://127.0.0.1:48530']' returned non-zero exit status 1

======================================================================
ERROR: test_uaread (tests_cmd_lines.TestCmdLines)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/python/Desktop/python-opcua-master/tests/tests_cmd_lines.py", line 36, in test_uaread
    s = subprocess.check_output(["python", "tools/uaread", "--url", self.srv_url, "--path", "0:Objects,4:directory,4:variable"])
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python', 'tools/uaread', '--url', 'opc.tcp://127.0.0.1:48530', '--path', '0:Objects,4:directory,4:variable']' returned non-zero exit status 1

======================================================================
ERROR: test_uawrite (tests_cmd_lines.TestCmdLines)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/python/Desktop/python-opcua-master/tests/tests_cmd_lines.py", line 40, in test_uawrite
    s = subprocess.check_output(["python", "tools/uawrite", "--url", self.srv_url, "--path", "0:Objects,4:directory,4:variable2", "1.789"])
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['python', 'tools/uawrite', '--url', 'opc.tcp://127.0.0.1:48530', '--path', '0:Objects,4:directory,4:variable2', '1.789']' returned non-zero exit status 1

----------------------------------------------------------------------
Ran 385 tests in 97.938s

FAILED (errors=4)

zerox1212, 2018-04-17:
This is for running python unit tests. Something in your environment is causing command line tools to have issues.

WangAllen, 2018-04-18:
how to solve these issues,could you help me