FreeOpcUa/python-opcua #580: Instancing object type with methods

MikelAyani, 2018-04-23:
Hi,
The code in the attached file shows that if an object type is created with a method, when instancing it, the method does not work.
[testerror.zip](https://github.com/FreeOpcUa/python-opcua/files/1940371/testerror.zip)
Regards,
Mikel

MikelAyani, 2018-04-24:
Adding this 2 ros after instancing the object makes it work:
    method = mydevice.get_methods()[0]
    mydevice.server.add_method_callback(method.nodeid, func)
Seems that the callback is not instantiated...
//Mikel

zerox1212, 2018-04-24:
There is a shortcut to do this `server.link_method()`. You can read the history here: https://github.com/FreeOpcUa/python-opcua/issues/297.

Basically it isn't fully implemented because we are not sure how the UA spec wants us to handle methods. Some people want the method to be static, others might want it to be an instance. For now you must link your desired method explicitly.

oroulet, 2018-04-24:
I think our implementation is correct

MikelAyani, 2018-04-24:
Thanks @zerox1212! I did not find the previous issue.
I understand it is not clear what to do with methods, but it would be great to have the option to select if you want to instantiate methods or not when creating new objects, like:
def add_object(self, nodeid, bname, objecttype=None, **instantiate_methods=False**):
With the solution you provide in #297 (but does not seem to be implemented) sounds easy to handle...
Thanks anyway for a great library!
//Mikel

oroulet, 2018-04-24:
We are not going to provide a special case for methods. They will be instantiated as everything else. You would have to have very good arguments to change that...

MikelAyani, 2018-04-24:
I just see methods as part of a class ([Object Model)](http://documentation.unified-automation.com/uasdkhp/1.0.0/html/_l2_ua_address_space_concepts.html), same as variables or properties, so for me sounds natural to instantiate all of them when creating a new object of a given ObjectType, I do not want to bother you so already modify a couple of things to make it work.
As said, thanks for your great effort!
//Mikel

closed 2018-04-24 by MikelAyani

zerox1212, 2018-04-24:
The UA side of the method is part of the object model and it does get instantiated, but the actual code of the method isn't. There is no place to in the model to store the python code for the corresponding method.

I will check what happened to `link_method()`. I thought that got merged a long time ago.