FreeOpcUa/python-opcua #581: Received a chunk with wrong token id 19, expected 20

helmut-jacob, 2018-04-24:
I'm seeing the following message a for a few seconds after the client renewed the secure channel.
`opcua.uaprotocol:Received a chunk with wrong token id 19, expected 20`

Reason is that PublishRequests from a client are queued on the server and when the server sends a PublishResponse it'll use the AlgorithmHeader of the original PublishRequest which contains the SecurityTokenId from before the renewal. Right now this just causes the warning to appear.

I guess the appropriate fix would be to generate the AlgorithmHeader when sending a response instead of re-using the header from the original request.

oroulet, 2018-04-24:
You are welcome to try...

helmut-jacob, 2018-04-24:
@oroulet do you remember why the implementation was done that way? Just to make sure we don't break existing functionality.

oroulet, 2018-04-24:
There was a lot of reverse engineering so that was probably just what we thought it should be

panadenbaer, 2018-09-20:
Hi,
got the same issue as @oroulet.

Im subscribing some vars and just put them into SQL table.

sometimes this happens

oroulet, 2018-09-20:
@helmut-jacob what about a PR since you looked at it?