FreeOpcUa/python-opcua #582: How to link/connect variables?

MikelAyani, 2018-04-24:
Hi,
I am sorry if this kind of questions should not be asked here, but I did not find any other place...
In my application, I would like to link or connect variables in a server, like produce/consumer variables. Basically, the value of a variable needs to be copied to the second one as fast as possible. I know I could do that by subscribing to the first variable and writing the value in the second one using a client. But is there any other way to do that? Maybe more effectively directly on the server?
I am also interested in connecting variables from different servers, something like mirroring, but I do not find information either...
Thanks in advance.
//Mikel

zerox1212, 2018-04-24:
I don't really understand your use case, but I will give you two possible options.

1. The server can subscribe to it's own nodes and your code can receive updates. You can look in the code and see this being done in the history module. You don't need a client to detect one update, and copy it to another.
2. You can interact directly with the node dict at the core of the server code if you can't wait for the asynchronous subscription updates, but I'm not sure how you would do it. If you search through old issues someone else was looking at doing this.

In the end I would make sure this problem can't be fixed with a better design. For example if every device was both a client and a server you wouldn't need to do mirroring.

MikelAyani, 2018-05-11:
Hi Zerox,
First of all, thanks for your answer and sorry for my late reply...
Basically, I am trying to use OPCUA to model a system for simulation, where objects are connected to each other. This means that some object outputs (variables) could be connected to other object inputs (variables). So on the server, I have created object instances with their I/Os and then other variables for describing the connections between I/Os (other variables). 
Now I am trying to find the fastest way to make these connections... I would like to make this updates in less than 1ms, but as far as I have been testing, I cannot update variables faster than 1.5-2ms... Even if the server and the client are running on the same computer...
I tried to find the option 2 you mention but couldn't...
Any other suggestion?
Thanks again!

oroulet, 2018-05-11:
I am not sure I understand but I can try to tell you to things: 1 you can subscribe to variables inside server, no need for a Client. 2 if you need performance under 1 ms then Python is the wrong language. Maybe try pypy..

oroulet, 2018-05-11:
Should be a bit faster without client. You avoid socket and many method calls

zerox1212, 2018-05-11:
Why do you need OPC UA to simulate two objects communicating? This use case doesn't make sense to me.