FreeOpcUa/python-opcua #585: USERNAME and PASSWORD LOGIN

drsso, 2018-05-04:
How  to  make server  that  allows  just  client   with  username and  PASSWORD to login 

I dosent find an example for it!!!!

drsso, 2018-05-06:
I try to implement it and I lost my way between the Lines. I am not a Python expert but I think I can do the JOB
**can you please tell me where to start in the CODE** cause its not easy to change too much in the source code
 uaprotocol_auto.py
            class UserNameIdentityToken(FrozenClass):
            class CreateSessionResponse(FrozenClass):
            
how do the client username and password will be sent to the server and how can I get them in the server side ??? in other form in witch class, function, or atrribute could i found them in the server implementetion and when this not implemented would you tell me where did the they have been thrown out from the client uri (in the code ) thereby i can get them

oroulet, 2018-05-06:
The user info is send when creating a session. Look into internal_server.py And search for where session is created. This is where you save user info and can deny session creation

oroulet, 2018-05-06:
But the main issue here is to decide how to store users and password safely. I have no idea. It would be great to use another (known secured) library and do not do things ourself in opcua. Place you code in user.py and replace the enumeration by a more advanced class

drsso, 2018-05-06:
my intention it to do it in users.py. i think of two ways to do it either to save them in a byte file or in sqlite.

my problem now is to implement:
1. create user
2. save users data ( i want to do it firstly in dictionary or list )
3. tell the client to enter user an pwd and check the authentecation -> make session

so you mean I have to make a class for authentication in users.py and then use it in **create_session(self, params, sockname=None):**

drsso, 2018-05-06:
please correct me if i dosnt understand it good
i have printed result in create_session in  internal_server.py but i didnt find my username that I gave to the client 
shouldn't username and pwd be sent with the connection request from client ????

oroulet, 2018-05-06:
This is supposed to work. Check parameters structure. Either create session or activate session

drsso, 2018-05-07:
now I could read the username and password from client and I check the connection with just a username and it works (simple test). my problem is now DECRYPTING PASSWORD  to be save it in a variable and play with it 
then :+1: 
- in users.py rewrite the Users enumuration class to set, get and check condition user data
- send login request to the client ... or check how it works 
- allow account creation deletion  just to Admin 
- save users data in file

oroulet, 2018-05-07:
Look at how password is encrypted in Client and do the same for decryption. But you will need to suport several encryption, at least no encryption

drsso, 2018-05-07:
i am a little confused 

i have decrypted it this way 
uacrypto.decrypt_rsa15(myprivateKEY,id_token.Password)

my password is **BOLD**  is there any way to take it out or am I on the wrong way ??

b',\x00\x00\x00**ABCDI999ABCD**\x98\xa2qN&\xa3.\x11F\xfb2\xaa\x84\xb9\xecy\xe6\xaf7\n\xf4}G\xfd\x00\xe8\x88\x13\x8b*\x84\xb3'

oroulet, 2018-05-07:
no idea what bold means...
be usre to support several encrytion and read it the correct one from parameters

drsso, 2018-05-08:
This  is  client  password: ABCDI999ABCD
And this  is what  I get  from  decrypting:

b',\x00\x00\x00ABCDI999ABCD\x98\xa2qN&\xa3.\x11F\xfb2\xaa\x84\xb9\xecy\xe6\xaf7\n\xf4}G\xfd\x00\xe8\x88\x13\x8b*\x84\xb3'

How  can  I deactivate  them Encription without Channing Evers things

FrankLin9981, 2018-07-17:
Hello,

I'm new to python OPC-UA.
I get stuck in username and pwd login.
I don't know how to start it.
Could you share some of codes?
If so, it will be a great help.
Sorry for bothering!

eclan, 2018-12-19:
Hello,

Is there an example for how to make server that allows client with username and PASSWORD to login?

zerox1212, 2018-12-19:
Some work was done on this, but I do not think an example was made.

You would have to look at https://github.com/FreeOpcUa/python-opcua/pull/691 and dig around in the code. I think there is a simple default user manager you can look at in the source.

If you can make an example and submit it would be great.

eclan, 2019-01-09:
Hello
I thank you for your answer.
I am writing a login example with an username and password. it works.
But I can not disable the Anonymous connection.
Here is my code :

```
iimport time

from opcua import ua, Server
from opcua.server.user_manager import UserManager

# users database
users_db = {
    'user1': 'passwd1',
    'user2': 'passwd2',
    'user3': 'passwd3',
}

# user manager
def user_manager(isession, username, password):
    print(isession, username, password)
    isession.user = UserManager.User
    return username in users_db and password == users_db[username]

if __name__ == "__main__":

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # load server certificate and private key. This enables endpoints
    # with signing and encryption.
    server.load_certificate("certificate-example.der")
    server.load_private_key("private-key-example.pem")

    # set all possible endpoint policies for clients to connect through
    server.set_security_policy([
        # ua.SecurityPolicyType.NoSecurity,
        ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
        # ua.SecurityPolicyType.Basic256Sha256_Sign,
    ])

    # set the security endpoints for identification of clients
    # self.server.set_security_IDs(["Anonymous", "Basic256Sha256", "Username"])
    server.set_security_IDs(["Username"])

    # set the user_manager function
    server.user_manager.set_user_manager(user_manager)

    # starting!
    server.start()

    print("Endpoints : ", str(server.get_endpoints()).replace(',', '\n'))

    try:
        while True:
            time.sleep(5)
    finally:
        # close connection, remove subscriptions, etc
        server.stop()

```

zerox1212, 2019-01-09:
It's likely that you can't disable it without editing the library. The general idea with this package is that connectivity is the focus, not security. Maybe another contributor knows if it's possible, but be aware that even the certificate is not fully enforced.

eclan, 2019-01-10:
OK. In the server.py file, it's written that : 
```
    def set_security_IDs(self, policyIDs):
        """
            Method setting up the security endpoints for identification
            of clients. During server object initialization, all possible
            endpoints are enabled:

            self._policyIDs = ["Anonymous", "Basic256Sha256", "Username"]

            E.g. to limit the number of IDs and disable anonymous clients:

                set_security_policy(["Basic256Sha256"])

            (Implementation for ID check is currently not finalized...)

        """
        self._policyIDs = policyIDs
```
I will try to see if i can do it.

okyame, 2019-01-11:
this issue was fix with #781