FreeOpcUa/python-opcua #586: calling a method with no output values

biberp, 2018-05-04:
Hi,

I am calling a method on a server that has two input arguments but no output argument. 
This fails in methods.pv line 44 with the following message:
result.OutputArguments = [var.Value for var in result.OutputArguments]
TypeError: 'NoneType' object is not iterable

Debugging this I have seen in my case result.OutputArguments is None in method call_method_full().
Checking the result.OutputArguments for None it fails in method call_method() again because of same problem missing check of result.OutputArguments.

Question: Do I miss use the method call? Is there an other way calling method with no output arguments? 

Thank you
Stefan

zerox1212, 2018-05-04:
Just make your output a single Boolean and return true when your method executes. You don't need to do anything with it...

carstone, 2019-03-26:
Hi, i encountered the same problem when using the link_method function from the opc-ua pakcage. It was a simple fix in my case, just return an ampty array [], from the python function. It is possible to have a function that doesent return anything, but when the function is made it by default will have an empty array as return paramter.