FreeOpcUa/python-opcua #591: unable to read data from Schneider PLC

gaofei207, 2018-05-11:
from opcua import Client
from opcua import ua

if __name__ == '__main__':
    # client=Client('opc.tcp://USER:USER@10.177.121.222:4840/')
    client=Client('opc.tcp://10.177.121.222:4840/')
    # try:
    client.connect()
    root=client.get_root_node()
    print(root.get_children())
    var = client.get_node("ns=2;s=POU.a1(%MW0)")
    print(var.get_data_value())

Error:
"opcua.ua.uaerrors._auto.BadNodeIdUnknown: The node id refers to a node that does not exist in the server address space.(BadNodeIdUnknown)"

If use UaExpert to test , I can read the value.
[
[image]
]
Is there something wrong in my code ?
Thank you in advance.

oroulet, 2018-05-11:
The error means the nice is not found. Most probably you have an error/typo in node id. With such a strange id you may have spaces somewhere or noon ASCII character.... Otherwise you will need to fire up Wireshark and record sessions

gaofei207, 2018-05-14:
@oroulet  Thanks very much.
 I  can get the data successfully like this.
 var = client.get_node("ns=2;i=1374562933").
But the id(1374562933) is from UaExerpt tool.
[image]

How can I get the nodeId via opcua library?
 root=client.get_root_node()
 print(root.get_children())
These only print out:
"[Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=87))]"

oroulet, 2018-05-14:
Client.noded.objectd.get_children() then call again get_children on every node you want. Or use browse name path

gaofei207, 2018-06-11:
I can read the variable successfully  now.  Thank you.
 var1=root.get_child(["0:Objects","2:M241-M251 data","2:POU.a1 (%MW0)"])
 print(var1.get_value()).
The reason for the previous mistake is that I missed a  blank space in browsename .

Thanks.

closed 2018-06-12 by gaofei207