FreeOpcUa/python-opcua #593: Writing a timestamp to variable

vanzylkruger, 2018-05-15:
Hi,

I can write values using set_value to a tag but is there a way of writing a timestamp to a tag.  We are writing results back the OPC UA server (kepserver) generated from a machine learning model and the time of calculation is not the same as the time it lands in the OPC UA server.  We would like to write the calculation time as the SourceTimeStamp of the tag?

oroulet, 2018-05-15:
Use get/set_data_value and pass a complete datavalue object

oroulet, 2018-05-15:
do a grep in example directory. or just call node.get_data_value(), inspect the object you get, change it and write it with set_data_value(dv)

vanzylkruger, 2018-05-15:
Thanks for your help so far.  I can't run my code locally because I have to deploy it to an environment where I have limited access so I can't access my code during execution, also there is no such example in the examples folder.  The documentation also doesn't cover this directly.  I have done an extensive search on the web on no luck.  

This works:

writeTag = client.get_node("ns=2;s="+opc_tag)
writeTag.set_value(ua.Variant(latest_egest_value, ua.VariantType.Float))

but for:

writeTag.set_data_value(?,?), what will it look like?

I will really appreciate your help, I am totally stuck.

zerox1212, 2018-05-15:
You are lucky I happened to have a screen shot of what the DataValue object looks like at runtime.
[image]

You can do `my_dv = node.get_data_value()` as mentioned above, edit `dv`, then call set.

Also, you don't need to run your specific code to solve this issue. You should really set up your local system to run a few of the examples in this library. `DataValue` is an object defined by the OPC UA spec. it isn't specific to user code.

vanzylkruger, 2018-05-16:
Getting a new error after using datavalue:

ERROR The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)

I really need to set the timestamp

zerox1212, 2018-05-22:
Sorry but this isn't the place to ask very basic Python questions solved by Googling and I don't blame him for being blunt. Like most open source libraries many of the people making issues here either:
A. Don't understand OPC UA and haven't read the specification or
B. Don't know basic Python
And lastly very few people actually contribute after we help them.

I think this repo is very helpful for people learning OPC UA, but Python support is better found else where. I mean you got 3 responses within 24 hours. Other repos I'm on you are lucky to get a 3 responses in a month.

vanzylkruger, 2018-05-22:
I am impressed by the rapid responses and I am grateful for that.  In the end it was not a Python issue.  I appreciate your efforts, we all have different styles.  I still feel there is more appropriate way of communicating, for instance, your response.  I am happy to to do (A) and (B) and I am more than happy to contribute when I can.

oroulet, 2018-05-22:
>> I have looked at some of your previous posts and they come across as unhelpful and rude
Well what responses were unhelpful? I do not have much time so if I can avoid answering it will help :-)