FreeOpcUa/python-opcua #596: call_method with multiple arguments

srep2512, 2018-05-16:
I'm trying to call a method with several arguments.
I use the public opcua server under this link.
opc.tcp://opcua.demo-this.com:51210/UA/SampleServer

I'm not sure how exactly the structure of the arguments must be defined to create a successful call

```
client = Client("opc.tcp://opcua.demo-this.com:51210/UA/SampleServer")
client.connect()
        parent = client.get_node("ns=2;i=10875")
        method = parent.get_child("0:AddComment")
        inputs = method.get_child("0:InputArguments").get_value()
        print("inputArgs are: ", inputs)        
        sas = str.encode("hello")
        arguments = [ua.Variant(stringTest,ua.VariantType.ByteString),ua.Variant("comment",ua.VariantType.String)]
        res = parent.call_method(method, *arguments)
```

Also, is there a way to generate code from the *InputArguments* in python, that i can use directly in the call method?

I would be grateful for a little help.

zerox1212, 2018-05-16:
What error do you get? There is nothing to generate code from input arguments. That would be something user specific.

zerox1212, 2018-05-16:
I tried doing the call from UAExpert on the public server you have in your comment and I can't get it to work. I dont' think the problem is the client.

Also, that method is asking for type LocalizedText. This is different than a String.

srep2512, 2018-05-17:
I think I still have problems with the data types. What would help is a little summary of how to define it in python.

thank you for your help :)