FreeOpcUa/python-opcua #606: Reading and writing data with opc ua

dalvor90, 2018-05-31:
Hi , I am a student who was doing a project in which I have to read some variables from the S7-1500 server. I have this code:
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
import sys
sys.path.insert(0, "..")
import time
import logging
import opcua
from IPython import embed
from opcua import Client, ua
from opcua.ua import ua_binary as uabin
from opcua.common.methods import call_method

logging.basicConfig(level=logging.WARN)

opc=Client("opc.tcp://192.168.1.1:4840")
Marcha=[]

opc.connect()

root = opc.get_root_node()

objeto=opc.get_objects_node()

Hijos= root.get_children()

Marcha_nodo= opc.get_node("ns=3;s=M_marcha")

Marcha=Marcha_nodo.get_value()

opc.disconnect()

This code is a start, surely I need many changes, but I do not know how to do it.
/////////////////////////////////////////////////////////////////////////////////////////////////////////////

I attach a UAExpert test client image

[image]

With this code it does not let me read the variable M_marcha, for example.
You can help me, I'm new to OPC UA and Python.
I need your help to finish my engineering studies.

Thank you very much.

oroulet, 2018-06-01:
Your code looks corect. What error do you get?

dalvor90, 2018-06-01:
The error is as follows.
[image]
I add an image of UAExpert connected to the server.

[image]

Thank you

dalvor90, 2018-06-01:
I have put and removed the quotes and I still give the same error :(

Thank you

oroulet, 2018-06-01:
you have some crazy nodeids here. Either you need to find exactly what character they used (And this can be hard because they may have some strange ones) or use path

    client.nodes.objects.get_child(["3:PLC", "0:Whaterver", etc....])

dalvor90, 2018-06-01:
Thanks, I've already managed to read. I went node to node until I reached the variable and it worked perfectly for me.
Adunjunto capture of the code.
[image]

gaofei207, 2018-06-06:
@dalvor90   hello, I am trying to read some variables from PLC too, but  failed. Could you please explain "3:PLC_1" and "3:Memmory"  in your last image? Or show a full screen of UaExpert? Thanks

andydqh, 2018-09-10:
what  server you used to connect SIMATIC PLC,