FreeOpcUa/python-opcua #608: Encryption and user authentication

kleskjr, 2018-06-03:
As far as i understood, clients can have two pairs or private key/certificate. The first one is set with load_private_key() and load_client certificate() methods and is used for authentication. The second pair is set with the set_security_string() method which are used for data decryption. Is this correct?

Currently, I am setting a list of trusted certificates in the server, so that the server can accept or decline clients during authentication. Now servers accept any clients, even with unmatching authentication key and certificate. So a veryfication of a matching private key is needed. It seems to be happening on server/uaprocessor.py Are there any thougths on this issue? Maybe someone of you have met this problems already?

oroulet, 2018-06-03:
The current situation is not surprising as I/we have been focusing on connecting to as many possible client/server not restrict it. But you proposition make sense

The client currently uses 2 certificates. one for authentication and one for encrytion. Thinking about it I do not really see why there are two certificates. it is probably possible to use the same for both I use the same for testing...BUT I am NOT a security expert and we should ask someone with more experience in that field before doing anything. Can you ask someone? @alkor any idea?

oroulet, 2018-06-04:
@kleskjr I am not sure when the verification should happen. but we should not a create a session if the verification failed, maybe put the code in iserver or in its own file..

kleskjr, 2018-06-04:
yes, verification should optimally happen before the connection is established. Currently I am using an ugly hack by veryfing the client signature, using the data certificate/key pair, but not the authentication ones. These days I would dig further to try for a more sane solution. In the mean time, should we think of introducing some kind of a variable/ list of trusted certificates? Or it might exist already?

kleskjr, 2018-06-07:
@oroulet I just pushed  _client_verification_ branch to my public repository. There, the signature of a client is veryfied during connection activation. I am not super deep into the crypto world, so it might be that I am missing something obvious or breaking some cryptotaboos. I would be happy if someone could have a look and help improving the verification process before doing a PR.

oroulet, 2018-06-07:
Just make a PR and start title with 'WIP:' so everybody understand this is work in progress