FreeOpcUa/python-opcua #609: get_value() method is not returning value of the node

AgrimaKothari1, 2018-06-05:
On using below code i am not able to get the value of particular node.
   client = Client("opc.tcp://<ip>/OPCUA/MyServer")
   client.connect()
   root = client.get_root_node()
   myvar = root.get_child(["0:Objects", "3:MyObjects", "3:MyMCRSystem" , "3:cpu0"])
   print(myvar.get_value())
print(myvar.get_data_value())

**Output received**: 
ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5003), Encoding:1, 24 bytes)
DataValue(Value:Variant(val:ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5003), Encoding:1, 24 bytes),type:VariantType.ExtensionObject), StatusCode:StatusCode(Good), SourceTimestamp:2018-06-05 05:07:48.106000)

On viewing it on OPCUA expert it looks like below (value is coming inside a box):
[image]
[image]

oroulet, 2018-06-05:
This is a server specific extension object. Call server.load_type_definition() first. I should add that call to example

AgrimaKothari1, 2018-06-05:
Thanks orulet, it worked with below code:
        client.connect()
        client.load_type_definitions() 
        struct = client.get_node("ns=3;i=5003")
        print(struct.get_value().cpu_usage_idle)