FreeOpcUa/python-opcua #612: ModifySubscription missing from server.

brubbel, 2018-06-05:
Hi,

It seems that support for the ModifySubscription service is not implemented in subscription_service.py.
According to the Part 7 of the OPC-UA specification, implementation is not optional to conform to the minimal 'Subscription Basic' profile (Part 7, rev 1.04, Table 10, page 13):

> SubscriptionBasic
> Support the following Subscription Services: CreateSubscription, ModifySubscription, DeleteSubscriptions, Publish, Republish and SetPublishingMode.

PS: this is not a request (not even a kind one) for implementation, but I want to report that at least one client (.NET) fails on this.

oroulet, 2018-06-05:
Can't you try to fix it? At least receiving the call and answering that everything is ok should be easy and make client happy

brubbel, 2018-06-05:
In this issue (https://github.com/FreeOpcUa/python-opcua/issues/594) you mentioned that some parts are autogenerated, is this also the case for subscription_service.py? Do you have a pointer for that?

oroulet, 2018-06-05:
I wrote that they could but it is not. Implementing it will be mainly copy paste from others and easy so try ;-)

closed 2018-06-05 by brubbel